dsmrreader/dsmr-reader #49: Use WeatherUnderground as weather service

jeroenpeters1986, 2016-02-13:
A friend of mine showed me Weather Underground, which is much more specific than the current source Buienradar. Almost every town has a use which shares his weather meter via this source.

I heard the API takes 150 free hits a day. I'm not sure that this is enough.

dennissiemensma, 2016-02-14:
Thanks for the suggestion! It seems very nice and might be a good addition for the dsmr_weather app. 

I think I recall this service from back when I was looking for weather API's. It requires the user to register and allows 500 calls per day. It also requires a full description of what to use it for. 
Should be easy for developers and depending on whether they validate all their free sign-ups, it might prove some worthsome data. This also happens to be the case for #9 and #10, so I think it should be possible.

I'll have to contact every party anyway to see if they have any trouble with the way/project their data is used/pushed for.

dennissiemensma, 2016-08-16:
One of the main reasons for this feature (for me) was to get more reliable weather information, to keep track of the temperatures combined with gas consumption. Mainly to compare winters.

In the meantime, several features have been added, such as #122 and #10 (which resolved #91). So I won't add this feature anymore. Also due to a lack of time available for the project. ;-)

closed 2016-08-16 by dennissiemensma