dsmrreader/dsmr-reader #67: Task scheduling alternative

dennissiemensma, 2016-02-29:
The backend currently consists of one big loop, feeded by signals, checking lots of stuff and being idle most of the time. As that works, in some way, it's not very effective.

Required for:
* Backup creation (now check every few seconds)
* Dropbox sync scheduling  (now check every few seconds)
* Weather readings (which now depends on gas reading compactor trigger)
* Day statistics compactor  (now check every few seconds)

It's important that there should be no more overhead as the application should perform on a raspberrypi 1 (but I'm thinking of abandoning it anyway) but **must** perform on a raspberrypi 2, without maxing the resources available.

danielterhorst, 2016-03-01:
You could look at Celery + Celerybeat + Redis. Redis can also be used for caching the Django app.

Redis's footprint is really low. Celery adds some; about the same as a Django app consumes. Celerybeat is quite a simple service so also barely adds anything. Got some good experience with this set-up now and the big advantage is that it works really easy with Django.

dennissiemensma, 2016-03-01:
Thanks, sounds good! Hereby noted to explore those options first. :]

dennissiemensma, 2016-11-07:
In the meantime (well actually on the same day when this ticket was creted) the RaspberryPi 3 was released, having lots of extra CPU power. 
This results in this ticket being either obsolete, as the current implementation should scale well enough, or more relevant than before. I won't however build this feature in the current stack so I'll close it for now.

I might combine it some day with a Docker implementation, should the need arise.

closed 2016-11-07 by dennissiemensma