dsmrreader/dsmr-reader #72: Source data retention (dsmr_datalogger_dsmrreading)

dennissiemensma, 2016-03-02:
This is my current database disk usage after almost three months, containing 700+ K readings:

```
 public.dsmr_datalogger_dsmrreading             | 104 MB
 public.dsmr_consumption_electricityconsumption | 15 MB
 public.dsmr_consumption_gasconsumption         | 512 kB
 public.dsmr_stats_hourstatistics               | 496 kB
 public.dsmr_weather_temperaturereading         | 184 kB
```

That averages around 400 MB for each year. It should be possible to delete source readings after:
* one month
* three months
* six months
* one year
* none (disabled)

Retention should be checked once a day or once a week.

dennissiemensma, 2016-03-24:
I thought of another solution. There is not need to dispose of all the source readings, as we could just wipe 99% of them in each hour. The absolute minimum for hourly stats is ofcourse at least one reading per hour.

If we would keep at least one reading of each hour passed, we could recalculate all statistics retroactive, if one needed to do so. As there are around 350+ readings in each hour, the data would be 350 times smaller. This should speedup backups, restores and migrations greatly. And reduce the disk space required.

The only downside is that the disk space freed might be reused (randomly) for new readings, possibly reducing the lifetime of the SD card. However, this would also be the situation when wiping all readings anyway.

dennissiemensma, 2016-11-20:
Update of my internal stats of disk used so far:

```
 public.dsmr_datalogger_dsmrreading             | 491 MB
 public.dsmr_consumption_electricityconsumption | 85 MB
 public.dsmr_stats_hourstatistics               | 1128 kB
 public.dsmr_consumption_gasconsumption         | 1064 kB
 public.dsmr_weather_temperaturereading         | 776 kB
 public.dsmr_stats_daystatistics                | 104 kB
```

Compared to March this year:

```
 public.dsmr_datalogger_dsmrreading             | 104 MB
 public.dsmr_consumption_electricityconsumption | 15 MB
 public.dsmr_consumption_gasconsumption         | 512 kB
 public.dsmr_stats_hourstatistics               | 496 kB
 public.dsmr_weather_temperaturereading         | 184 kB
```

As I'm almost reaching the three million milestone DSMR-reading count: `2.973.738`

dennissiemensma, 2016-11-24:
I've looked into this for some hours now, but it's quite complex to implement semi-retention for discarding data, while still keeping 1 data point per hour. I'll won't have this delay the 1.4 release, so next try in v1.5.

dennissiemensma, 2016-11-27:
Note to self to also update the docs, as I refer to this feature multiple times.

dennissiemensma, 2016-12-08:
Since `v1.4` was released, I worked on the Django upgrade for `v1.5`. This added some nice featured Django lacked before. I'm now able to query *exactly* what I'm looking for, without hacks.

Django's new `TruncHour` ORM function allows me to extract the hour of each reading (by truncating the 'minutes' and 'seconds' to zero). Then I group them, summarize their count per hour and filter out any hours which only (or already) have one reading left.

```
DsmrReading.objects.filter(
    timestamp__lt=retention_date
).annotate(
    item_hour=TruncHour('timestamp')
).values('item_hour').annotate(
    item_count=Count('id')
).order_by().filter(
    item_count__gt=1
)
```

This is so nice as it allows me to retroactively iterate through any data requiring retention. And I can limit that as well to prevent any hanging background processes.

dennissiemensma, 2016-12-13:
Committed an initial setup in 54697945d346fb821b5f499277458d497bae6123. To be continued.

dennissiemensma, 2016-12-19:
Moved to `v1.6` release, as I hit some trouble with MySQL + timezones.

dbmnl, 2017-07-14:
Hi Dennis,

Just an FYI to you regarding my DB usage at the moment because they are different from your measurements.

total amount of measurments according to the statistics page: 1.851.825

Size according to \l+

[image]

If I compare that to your 2.9 mil entries there is a big difference. So i checked the tables itself for reference if this was true, but noticed that the added total can never be the amount of MB's reported in the image above.

public     | dsmr_datalogger_dsmrreading                     | table    | dsmrreader | 183 MB 
public     | dsmr_consumption_electricityconsumption         | table    | dsmrreader | 162 MB 
public     | dsmr_stats_daystatistics                        | table    | dsmrreader | 48 kB 
 public     | dsmr_stats_hourstatistics                       | table    | dsmrreader | 376 kB public     | dsmr_weather_temperaturereading                 | table    | dsmrreader | 264 kB

Any Idea why there is such a difference between the two and also which one should be treated as "the truth"?

dennissiemensma, 2017-07-14:
The 'truth' is what your filesystem is reporting to you. And the data size is the size of your backup, as it does not contain the disk space of any indexes created by Postgres.

My current database size is:

```
                    relation                    | total_size 
------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading             | 810 MB
 public.dsmr_consumption_electricityconsumption | 193 MB
```

The actual diskspace (MB) is:
```
 du -msc /var/lib/postgresql/
1772	/var/lib/postgresql/
```

The number of readings stored: `4.952.599`

I'm not sure why there is a difference. Also, my stats are (by coincidence) a clean imported database, as my SD card happened to crash this week and I had to restore a backup. 

The raw data as stored, in my latest backup, is `~655 MB`. So I guess this is just how Postgres works.

dbmnl, 2017-07-15:
Clear, tnx for the answer dennis. Just FYI these are the statistics a day later.

```
\l+
875 MB DB size 

```

```
\dS+
dsmr_consumption_electricityconsumption 162 MB
dsmr_datalogger_dsmrreading 184 MB

``````
du -msc /var/lib/postgresql/
961     /var/lib/postgresql/
```

With 1.860.070 entries

dennissiemensma, 2017-10-12:
Inmiddels heb ik hier een feature voor kunnen ontwikkelen. Deze gooit simpelweg van bij metingen en elektra-/gasverbruik alles weg, behalve één item per uur.

Voor een DSMR v4 meter betekent dat dus ~359 van de 360 metingen (per uur) verwijderd worden. Bij een DSMR v5 meter ligt dat nog tot 10 maal zo hoog.

Dit is mijn huidige database grootte:
```
 public.dsmr_datalogger_dsmrreading              | 926 MB
 public.dsmr_consumption_electricityconsumption  | 221 MB
 public.dsmr_stats_hourstatistics                | 1816 kB
 public.dsmr_consumption_gasconsumption          | 1688 kB
 public.dsmr_weather_temperaturereading          | 1496 kB
 public.dsmr_stats_daystatistics                 | 160 kB
```

En dit is na retentie op 24 uur + `vacuumdb -a -f` (nodig voor postgres om schuifruimte vrij te geven):

```
 public.dsmr_datalogger_dsmrreading              | 4704 kB
 public.dsmr_consumption_electricityconsumption  | 4416 kB
 public.dsmr_stats_hourstatistics                | 1792 kB
 public.dsmr_consumption_gasconsumption          | 1664 kB
 public.dsmr_weather_temperaturereading          | 1472 kB
 public.dsmr_stats_daystatistics                 | 136 kB
```

En zo zien de metingen er naderhand uit. Van elk uur er is tenminste 1 (de eerste) bewaard:

```
------------------------
 2015-12-11 22:25:05+01
 2015-12-11 23:00:05+01
 2015-12-12 00:00:02+01
 2015-12-12 01:00:09+01
 2015-12-12 02:00:06+01
 2015-12-12 03:00:04+01
 2015-12-12 04:00:01+01
 2015-12-12 05:00:08+01
 2015-12-12 06:00:05+01
 2015-12-12 07:00:03+01
 2015-12-12 08:00:00+01
 2015-12-12 09:00:07+01
 2015-12-12 10:00:04+01
 2015-12-12 11:00:02+01
 2015-12-12 12:00:09+01
 2015-12-12 13:00:06+01
 2015-12-12 14:00:03+01
 2015-12-12 15:00:01+01
 2015-12-12 16:00:08+01
 2015-12-12 17:00:05+01
 2015-12-12 18:00:02+01
 2015-12-12 19:00:00+01
 2015-12-12 20:00:07+01
 2015-12-12 21:00:04+01
 2015-12-12 22:00:01+01
 2015-12-12 23:00:09+01
 2015-12-13 00:00:06+01
 2015-12-13 01:00:03+01
 2015-12-13 02:00:00+01
 2015-12-13 03:00:08+01
 2015-12-13 04:00:05+01
 2015-12-13 05:00:02+01
 2015-12-13 06:00:09+01
 2015-12-13 07:00:07+01
 2015-12-13 08:00:04+01
 2015-12-13 09:00:01+01
 2015-12-13 10:00:08+01
 2015-12-13 11:00:06+01
 2015-12-13 12:00:03+01
```

En idem voor electriciteitsverbruik (en gas, wanneer er met DSMR v5 meer dan 1 meting per uur is):
```
 2015-12-11 22:26:00+01
 2015-12-11 23:00:00+01
 2015-12-12 00:00:00+01
 2015-12-12 01:00:00+01
 2015-12-12 02:00:00+01
 2015-12-12 03:00:00+01
 2015-12-12 04:00:00+01
 2015-12-12 05:00:00+01
 2015-12-12 06:00:00+01
 2015-12-12 07:00:00+01
 2015-12-12 08:00:00+01
 2015-12-12 09:00:00+01
 2015-12-12 10:00:00+01
 2015-12-12 11:00:00+01
 2015-12-12 12:00:00+01
 2015-12-12 13:00:00+01
 2015-12-12 14:00:00+01
 2015-12-12 15:00:00+01
 2015-12-12 16:00:00+01
 2015-12-12 17:00:00+01
 2015-12-12 18:00:00+01
 2015-12-12 19:00:00+01
 2015-12-12 20:00:00+01
 2015-12-12 21:00:00+01
```

Van de 5,5+ miljoen metingen, zijn er hier nog een schamele 15.000+ over. Tenminste, op een kopie van mijn productieomgeving.

dennissiemensma, 2017-10-12:
Wat nog niet werkt is het compleet hergenereren van de dagtotalen op basis van de overgebleven gegevens. Daar komen redelijk grote verschillen uit, dus ik zal daar naar moeten kijken.

dennissiemensma, 2018-01-02:
Gezien DSMR v5, en de grote hoeveelheid gegevens, voor uitdagingen zorgt, zal ik hier mee verder gaan. 

Wellicht dat ik het verlies van nauwkeurigheid maar accepteer (bij bewaren van 1 meting per uur), omdat het vermoedelijk redelijk wat tijd kost om uit te vogelen hoe het hergenereren van gegevens zo kan afwijken. Zolang men backups heeft en ze extern bewaart, blijven de dagstatistieken van eerdere jaren/maanden altijd beschikbaar.

dennissiemensma, 2018-01-10:
Eerste opzet is klaar. Het werkt goed voor PostgreSQL. Helaas nog niet voor MySQL, dus daar mag ik nog een blik op werpen.

dennissiemensma, 2018-01-11:
Bij nader inzien werkt het wel voor MySQL, maar alleen wanneer de tijdzones geactiveerd zijn in de database én de tijdzone op UTC staat in de MySQL config.

Wellicht exclude ik deze functionaliteit simpelweg voor MySQL.

dennissiemensma, 2018-01-11:
Dit zou afdoende moeten zijn voor degenen die nog MySQL gebruiken: `mysql_tzinfo_to_sql /usr/share/zoneinfo | mysql mysql` (als root)

dennissiemensma, 2018-01-13:
Branch gemerged naar `development`. Ik zal nog wat laatste zaken controleren en daarna wil ik de pre-release erg zelf een dagje draaien met retentie. 
Kijken of de Raspberry/DB het uberhaupt aan kan qua performance. Op mijn ontwikkel-PC draait het namelijk, vanzelfsprekend, goed.

dennissiemensma, 2018-01-13:
Uitgerold. De performance is nog niet super, maar ik kan daar weinig aan veranderen. Wellicht dat ik het doorvoeren van retentie laat toepassen voor bepaalde periodes's nachts, zodat andere processen er ook geen last van hebben.

```
2018-01-13 13:09:24.282502+01:00: Calling backend services
Retention | Cleaning up: 2015-12-11 21:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-11 22:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-11 23:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 00:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 01:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 02:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 03:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 04:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 05:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 06:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 07:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 08:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 09:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 10:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 11:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 12:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 13:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 14:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 15:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 16:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 17:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 18:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 19:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-12 20:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2015-12-11 21:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-11 22:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-11 23:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 00:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 01:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 02:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 03:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 04:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 05:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 06:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 07:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 08:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 09:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 10:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 11:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 12:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 13:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 14:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 15:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 16:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 17:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 18:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 19:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2015-12-12 20:00:00+00:00 (ElectricityConsumption)
 - Processed reading: 6498150 @ 2018-01-13 13:08:43+01:00 (0.156 kW)
 - Processed reading: 6498151 @ 2018-01-13 13:08:53+01:00 (0.156 kW)
 - Processed reading: 6498152 @ 2018-01-13 13:09:03+01:00 (0.155 kW)
 - Processed reading: 6498153 @ 2018-01-13 13:09:13+01:00 (0.154 kW)
 - Processed reading: 6498154 @ 2018-01-13 13:09:23+01:00 (0.157 kW)
 - Processed reading: 6498155 @ 2018-01-13 13:09:33+01:00 (0.165 kW)
 - Processed reading: 6498156 @ 2018-01-13 13:09:43+01:00 (0.169 kW)
Command completed. Sleeping for 1 second(s)...
```

Hij heeft nu op een Pi 3 ongeveer een minuut nodig per dag opschoning. En dat is nog maar voor een relatief kleine database.

dennissiemensma, 2018-01-13:
Voor nu zal ik hem bij mijzelf aan laten staan, zodat ik kan zien hoelang het duurt om een jaar aan gegevens op te schonen. Ter referentie mijn huidige database:

```
                    relation                    | total_size 
------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading             | 1159 MB
 public.dsmr_consumption_electricityconsumption | 266 MB
 public.dsmr_stats_hourstatistics               | 2064 kB
 public.dsmr_consumption_gasconsumption         | 1928 kB
 public.dsmr_weather_temperaturereading         | 1712 kB
 public.dsmr_stats_daystatistics                | 168 kB
```

dennissiemensma, 2018-01-13:
Ik heb nu een hardcoded restrictie toegevoegd dat retentie alleen tussen 0:00 en 6:00 uitgevoerd wordt (wanneer ingeschakeld). Wellicht dat ik het later nog instelbaar maak, maar het heeft weinig zin retentie overdag door te voeren, gezien het veel resources kost (in het begin).

Vanmiddag heeft mijn eigen Pi zo'n 6 uur lang retentie kunnen doorvoeren. Van de `6,4M` metingen zijn er inmiddels nog zo'n `3,4M` over. 

Gezien PostgreSQL zelf niet altijd loze ruimte vrijgeeft, doe ik nu handmatig een `vacuumdb` om te kijken wat het oplevert:

```
INFO:  vacuuming "public.dsmr_consumption_electricityconsumption"
INFO:  "dsmr_consumption_electricityconsumption": found 63783 removable, 619511 nonremovable row versions in 11737 pages
DETAIL:  0 dead row versions cannot be removed yet.
CPU 0.53s/2.10u sec elapsed 18.26 sec.

INFO:  vacuuming "public.dsmr_consumption_gasconsumption"
INFO:  "dsmr_consumption_gasconsumption": found 0 removable, 18116 nonremovable row versions in 134 pages
DETAIL:  0 dead row versions cannot be removed yet.
CPU 0.03s/0.02u sec elapsed 0.47 sec.

INFO:  vacuuming "public.dsmr_datalogger_dsmrreading"
INFO:  "dsmr_datalogger_dsmrreading": found 2370 removable, 3619955 nonremovable row versions in 84298 pages
DETAIL:  0 dead row versions cannot be removed yet.
CPU 3.23s/15.20u sec elapsed 103.58 sec.
```

Met als concreet resultaat:
```
------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading             | 607 MB
 public.dsmr_consumption_electricityconsumption | 147 MB
 public.dsmr_stats_hourstatistics               | 2040 kB
 public.dsmr_consumption_gasconsumption         | 1904 kB
 public.dsmr_weather_temperaturereading         | 1688 kB
 public.dsmr_stats_daystatistics                | 160 kB
```

dennissiemensma, 2018-01-13:
Aankomende nacht heeft die dus opnieuw 6 uur de tijd om het restant op te schonen. Waarschijnlijk is die daarna helemaal bij. Morgen opnieuw kijken.

dennissiemensma, 2018-01-14:
Vannacht is het restant opgeschoond. Er zijn, van de eerdere 6 miljoen metingen, nu nog maar een kleine 300.000 over.

Een vacuumdb schoont de laatste ruimte op en

Disk usage was:
```
                    relation                    | total_size 
------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading             | 1159 MB
 public.dsmr_consumption_electricityconsumption | 266 MB
 public.dsmr_stats_hourstatistics               | 2064 kB
 public.dsmr_consumption_gasconsumption         | 1928 kB
 public.dsmr_weather_temperaturereading         | 1712 kB
 public.dsmr_stats_daystatistics                | 168 kB
```

Nu:
```
                   relation                     | total_size 
-------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading              | 47 MB
 public.dsmr_consumption_electricityconsumption  | 18 MB
 public.dsmr_stats_hourstatistics                | 2048 kB
 public.dsmr_consumption_gasconsumption          | 1904 kB
 public.dsmr_weather_temperaturereading          | 1688 kB
 public.dsmr_stats_daystatistics                 | 160 kB
```

Dat betreft dus 2 jaar aan metingen, sinds december 2015, waarbij de rententie momenteel op max één maand staat. De database-backup is nu ook lekker klein en vooral snel.

Bij gebruikers met een DSMRv5-meter zullen de verschillen vanzelfsprekend nóg groter zijn.

dennissiemensma, 2018-01-14:
Ik zal nog een howto voor `vacuumdb` toevoegen aan de docs, plus een waarschuwing van verliezen nauwkeurigheid bij hergenereren gegevens.

closed 2018-01-14 by dennissiemensma

dennissiemensma, 2018-01-14:
Zojuist uitgebracht in [release v1.12](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.12.0).

mkruiver, 2018-01-14:
Vraagje mbt regenereren van statistieken: je gaf aan dat zolang men externe backups toepast de statistieken altijd beschikbaar blijven, maar, je kunt backups niet stapelen bij het restoren. Hoe gaat dit dan in zn werk?

dennissiemensma, 2018-01-14:
Ik doel daarbij op de dag- en uurstatistieken-tabellen die nooit opgeschoond zullen worden. Zolang je goede database backup's hebt, zul je die altijd kunnen restoren, ongeacht tot hoever de bron-metingen opgeschoond zijn.
Het enige nadeel daarbij is wel dat je nooit je dag- en uurstatistieken (eenvoudig) opnieuw kan genereren met bijvoorbeeld een andere prijs (als je dat om wat voor reden dan ook zou willen).

Een alternatief is **geen** retentie inschakelen en het handmatig beperken van de doorvoer aan metingen. Zie daarvoor #413. Alleen zal dat niet voor iedereen een werkbare/wenselijke situatie zijn.

mkruiver, 2018-01-15:
Ik heb de retentie op één maand gezet. Vannacht tussen 1 en 7 (dus niet tussen 0 en 6!) heeft deze duidelijk gedraaid want de PVO output was bijna non-existent.

Raadt je aan om pas nadat het opruimen klaar is te vacuum-en? En moeten we dan een vacuum FULL ANALYZE doen?

dennissiemensma, 2018-01-15:
Ik zou persoonlijk de vacuum pas doen wanneer hij helemaal klaar is. Het is namelijk een proces dat volgens mij de complete database vergrendelt en bij mij had die el 5 a 10 minuten nodig, laat staan bij een grote database zoals die van jou.

Je kunt overigens in de logs van de backend vrij makkelijk zien waar die is gebleven én hoelang het duurt om één dag op te schonen (bij mij iets meer dan een minuut).

Als je zoekt op `Retention` in `/var/log/supervisor/dsmr_backend.log` zou je zoiets moeten zien:

```
2018-01-15 01:00:00.402471+01:00: Calling backend services
Retention | Cleaning up: 2017-12-17 07:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 08:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 09:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 10:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 11:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 12:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 13:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 14:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 15:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 16:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 17:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 18:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 19:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 20:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 21:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 22:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 23:00:00+00:00 (ElectricityConsumption)
Retention | Cleaning up: 2017-12-17 07:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 08:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 09:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 10:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 11:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 12:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 13:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 14:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 15:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 16:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 17:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 18:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 19:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 20:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 21:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 22:00:00+00:00 (DsmrReading)
Retention | Cleaning up: 2017-12-17 23:00:00+00:00 (DsmrReading)
```

Tot slot, qua type vacuum, ik heb tot nu toe gewoon een FULL gedaan, maar je kunt kijken of dat uberhaupt noodzakelijk is.

mkruiver, 2018-01-19:
Naar mijn mening is de beschrijving van deze feature in de docs niet helemaal correct. Je schrijft over het "inschakelen van retentie" terwijl de default ('rententie uit') juist alle gegevens bewaart. Deze beschrijving is eigenlijk verkeerd om.

mkruiver, 2018-01-19:
Doet overigens niks af aan de functie zelf, dit is een zeer nuttige toevoeging!

dennissiemensma, 2018-01-19:
Bedankt voor je input. Ik heb het vooral beschreven vanuit het technisch oogpunt, echter klopt het dat het op meerdere manieren gelezen kan worden. Daarom heb ik het nu iets aangepast en mogelijk mag ik het nog verder herschrijven een volgende keer.

mkruiver, 2018-01-19:
Maybe a rename of the feature can solve this.. Retention Reduction or
something.