dsmrreader/dsmr-reader #73: Datalogger doesn't work properly with DSMR 4.2

danielterhorst, 2016-03-03:
Example 4.2 telegram of my energy reader below.

Error details:
- https://sentry.dth.wtf/share/issue/322e34/
- https://sentry.dth.wtf/share/issue/322e36/

```
/KFM5KAIFA-METER

1-3:0.2.8(42)
0-0:1.0.0(160303164347W)
0-0:96.1.1(*******************************)
1-0:1.8.1(001073.079*kWh)
1-0:1.8.2(001263.199*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.143*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(000101000001W)(2147483647*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(000*A)
1-0:21.7.0(00.143*kW)
1-0:22.7.0(00.000*kW)
!74B0
```

dennissiemensma, 2016-03-03:
Thanks! I'll look into it.

dennissiemensma, 2016-03-03:
I think it's caused by these fields, which do not seem to appear on your meter. I asumed the DSMR version determined the avalability of these fields, but appearently the meter's manufacturer decides whether these are available.

```
    # Static data, stored in database but only record of the last reading is preserved.
    '1-0:32.32.0': 'voltage_sag_count_l1',
    '1-0:52.32.0': 'voltage_sag_count_l2',
    '1-0:72.32.0': 'voltage_sag_count_l3',
    '1-0:32.36.0': 'voltage_swell_count_l1',
    '1-0:52.36.0': 'voltage_swell_count_l2',
    '1-0:72.36.0': 'voltage_swell_count_l3',
```

Good thing is that they are not required for the datalogger. I separated them recently, and made them optional, but that does not seem to work somehow (stored in `MeterStatistics` model). 

I'll add your snapshot to the tests and hopefully it will give me more info.

dennissiemensma, 2016-03-03:
Thankfully it crashes the test as well:

```
======================================================================
ERROR: Test whether dsmr_datalogger can insert a reading for Kaifa DSMR v4.2.
----------------------------------------------------------------------
...
    statistics_kwargs = {k: parsed_reading[k] for k in _get_statistics_fields()}
KeyError: 'voltage_sag_count_l2'
```

Relays to the same crash as your Sentry dump. (Which is a nice addition to #65)

dennissiemensma, 2016-03-03:
Fixed it. Also explains why the NULLable fields didnt work. The reading dict (which is passed on to the database) itself wasn't defaulting to None for the extra fields mentioned above.

Old:
```
    # Defaults all fields to NULL.
    parsed_reading = {k: None for k in _get_reading_fields()}
```

New:
```
    # Defaults all fields to NULL.
    parsed_reading = {k: None for k in _get_reading_fields() + _get_statistics_fields()}
```

dennissiemensma, 2016-03-03:
I'll complete the tests for your snapshot and will merge it after.

dennissiemensma, 2016-03-03:
Done. Branch passed build so merged: https://travis-ci.org/dennissiemensma/dsmr-reader/jobs/113469409

dennissiemensma, 2016-03-03:
The latest master branch should fix your problem. Please reopen issue when the fix does not seem to work. Once again thanks for reporting :]

closed 2016-03-03 by dennissiemensma