dsmrreader/dsmr-reader #77: Archive only shows graph untill 23:00 (11 pm)

danielterhorst, 2016-03-04:
The graphs on the archive page are only displaying data until 23:00 (11 pm). The last hour has fallen off :-)

dennissiemensma, 2016-03-04:
Thanks for reporting. I've looked into this and it does seem to happen, but only when the **last** day (yesterday) is displayed, which happens to be the default. If you lookup any days before, then it should display 0:00 to 0:00.

Archive view:
```
HourStatistics.objects.filter(
            hour_start__gte=selected_datetime,
            hour_start__lte=selected_datetime + timezone.timedelta(days=1)
        )
```

However, that does not explain why the hourly stats are missing. I seem to have a hole in my own hourly stats for `22:00:00+00`, every day. I will look into this later, as it seems data compacting fails, possible also causing the 23:00 - 0:00 gap.

dennissiemensma, 2016-03-15:
The 'next' midnight hour is never called, because of `for current_hour in range(0, 24):`, which skips the `24`.

I'll have to redesign `create_hourly_statistics()` to allow this.

dennissiemensma, 2016-03-15:
I will test the possible solution branched on my own production first for tonight.

dennissiemensma, 2016-03-16:
The issue is resolved, as I now see midnight at the end of the day. 

Last remaining question is whether the possible integrityerror thrown for duplicate hours is caught as expected. That should happen tonight, so I'll merge tomorrow if the issue is resolved.

dennissiemensma, 2016-03-17:
Too bad, it crashed or something, as I am missing the statistics for today's hours.

dennissiemensma, 2016-03-17:
Wall of text, it only crashes the backend process once, as the day totals are generated first.

```
- <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_stats>> ::   File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 225, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/apps.py", line 20, in _on_backend_called_signal
    dsmr_stats.services.analyze()

  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/services.py", line 92, in analyze
    create_hourly_statistics(hour_start=hour_start)

  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/services.py", line 127, in create_hourly_statistics
    if not electricity_readings.exists():

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/query.py", line 586, in exists
    return self.query.has_results(using=self.db)

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/query.py", line 484, in has_results
    return compiler.has_results()

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 811, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 840, in execute_sql
    cursor.execute(sql, params)

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/utils.py", line 79, in execute
    return super(CursorDebugWrapper, self).execute(sql, params)

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/utils.py", line 59, in execute
    self.db.validate_no_broken_transaction()

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 327, in validate_no_broken_transaction
    "An error occurred in the current transaction. You can't "
  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 225, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/apps.py", line 20, in _on_backend_called_signal
    dsmr_stats.services.analyze()

  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/services.py", line 92, in analyze
    create_hourly_statistics(hour_start=hour_start)

  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/services.py", line 127, in create_hourly_statistics
    if not electricity_readings.exists():

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/query.py", line 586, in exists
    return self.query.has_results(using=self.db)

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/query.py", line 484, in has_results
    return compiler.has_results()

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 811, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 840, in execute_sql
    cursor.execute(sql, params)

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/utils.py", line 79, in execute
    return super(CursorDebugWrapper, self).execute(sql, params)

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/utils.py", line 59, in execute
    self.db.validate_no_broken_transaction()

  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 327, in validate_no_broken_transaction
    "An error occurred in the current transaction. You can't "
CommandError: ['  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 225, in send_robust\n    response = receiver(signal=self, sender=sender, **named)\n\n  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/apps.py", line 20, in _on_backend_called_signal\n    dsmr_stats.services.analyze()\n\n  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/services.py", line 92, in analyze\n    create_hourly_statistics(hour_start=hour_start)\n\n  File "/home/dennis/workspace/dsmr-reader/dsmr_stats/services.py", line 127, in create_hourly_statistics\n    if not electricity_readings.exists():\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/query.py", line 586, in exists\n    return self.query.has_results(using=self.db)\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/query.py", line 484, in has_results\n    return compiler.has_results()\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 811, in has_results\n    return bool(self.execute_sql(SINGLE))\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 840, in execute_sql\n    cursor.execute(sql, params)\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/utils.py", line 79, in execute\n    return super(CursorDebugWrapper, self).execute(sql, params)\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/utils.py", line 59, in execute\n    self.db.validate_no_broken_transaction()\n\n  File "/home/dennis/.virtualenvs/dsmr-reader/lib/python3.4/site-packages/django/db/backends/base/base.py", line 327, in validate_no_broken_transaction\n    "An error occurred in the current transaction. You can\'t "\n']
```

dennissiemensma, 2016-03-17:
It seems that an integrity error breaks any transaction, disregarding whether the exception is caught somewhere in the process. I've decided to let hour statistics do nothing when data already exists.

dennissiemensma, 2016-03-17:
Should be fixed now.

closed 2016-03-17 by dennissiemensma

dennissiemensma, 2016-03-18:
I've check it today and I can now see data until midnight of the previous day. Please reopen if the fix does not resolve your problem. :]