dsmrreader/dsmr-reader #101: Monitor requirements with requires.io

dennissiemensma, 2016-03-22:
Requires.io (https://requires.io/) keeps track of all dependencies and whether they are outdated. Seems fair enough to try.

dennissiemensma, 2016-03-22:
It seems some dependencies are outdated already: https://requires.io/github/dennissiemensma/dsmr-reader/requirements/?branch=master

dennissiemensma, 2016-03-22:
I've updated `django-solo`, `dropbox`, `gunicorn`, `python-dateutil` and `pytz`. I will not update `Django` as current version `1.8` is LTS, which is fine for me (and this project).

dennissiemensma, 2016-03-22:
Tests pass, but before merging I'll deploy and test Gunicorn on production just to be sure.

dennissiemensma, 2016-03-22:
It works, also restarted Gunicorn with a hard restart. Dropbox was updated but sync still seems to work fine.

closed 2016-03-22 by dennissiemensma

jeroenpeters1986, 2016-03-22:
Cool tool :) Authored by the same team that wrote ShiningPanda, the clean-virtual-env builder :D

dennissiemensma, 2016-03-22:
Yeah it's nice, but the only downside is that it marks Django LTS as outdated (even the latest patch). Well, can't have it all :]

jeroenpeters1986, 2016-03-22:
Difference of opinion/vision I think. Cutting edge vs long term support.
Well, as you say: can't have it all :)