dsmrreader/dsmr-reader #125: Trends page giving errors

WatskeBart, 2016-05-01:
Following error when navigating to /trends
Running release v1.1.1
[image]

dennissiemensma, 2016-05-01:
Hi, thank you for reporting this issue. This will happen when there is no data (yet) to plot trends with.

Could you provide some more information such as:

**A snapshot of one your DSMR telegrams (readings)**

- Either with `cu` (http://dsmr-reader.readthedocs.io/en/latest/installation.html#your-first-reading-optional)

- Or just tail the logfile of the datalogger process: `sudo supervisorctl tail -f dsmr_datalogger` (http://dsmr-reader.readthedocs.io/en/latest/installation.html#supervisor)

You should see some data similar to this format: https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/tests/datalogger/test_landisgyr350_dsmr42.py#L17

**What kind of data does your Status-page provide?**
Any errors and does your reading cont update? (http://dsmr-reader.readthedocs.io/nl/latest/_static/screenshots/status.png)

WatskeBart, 2016-05-01:
I already suspected that the error was because of lack of data. All other pages are working OK and data is being received.

Maybe implement a check to see if enough data is generated and telling the user there isn't enough data yet to make a trend. The current error could make a user think they did something wrong.
Something like you did on the Archive page:
[image]

Still what a great application you wrote. I tried too many bad dsmr implementations before I discovered your application. Good stuff, keep it up

dennissiemensma, 2016-05-01:
Thank you for your feedback and kind words. 

There used to be a check but I suspect the changes for issue #99 (a while back) make it crash before the GUI could render. I will fix the crash as it shouldn't behave in this manner.

Feel free to report other issues or ideas as well. :]

dennissiemensma, 2016-05-01:
Fixed and added test in commit f2cbbd9. Will be released shortly as patch.

closed 2016-05-01 by dennissiemensma

dennissiemensma, 2016-05-01:
Patch is available as version `1.1.2`. 

- You may update codebase & reload manually.
Or
- Just use the ``./deploy.sh`` script in the root of this project (run as `dsmr` user).

Also updated credits: http://dsmr-reader.readthedocs.io/en/latest/credits.html

WatskeBart, 2016-05-01:
Just updated and looking good!