dsmrreader/dsmr-reader #130: Tariff selection

WatskeBart, 2016-05-02:
Energy suppliers offer two kinds of tariffs, namely;

1. Low-tariff / High-tariff
2. Single-tariff

Maybe implement a selection in the **/admin/dsmr_consumption/consumptionsettings/** section, to choose between the two kinds of tariffs.
Also when a single-tariff is selected, only a single-tariff will be displayed in the **/admin/dsmr_consumption/energysupplierprice/** section.

This way people don't have to enter the same price per kWh in the low and high-tariff section.
Maybe implement it on the frontend something like, Electricity 1+2 (single-tariff)

dennissiemensma, 2016-05-02:
Thanks for the suggestion! 

The current data format does not allow (dynamic) single tariff, because of the way the data modeled and Django's (framework) admin interface works. 
It might be possible, but will need quite some adjustments in:

- Add an option in `EnergySupplierPrice` for enforce this 'mode'.
- `EnergySupplierPrice` model, having a third field for a single price (or mask one of the two existing ones).
- Hack the Django admin interface to display fields depending on the setting of another one.
- Alter logic in `day_consumption()` service to take care of this mode.
- Possibly combine of check all frontend locations of pricing.

Therefor I will push this request back for a possible release later this year, simply because it will require quite some refactoring. Still a nice idea though.

dennissiemensma, 2016-11-17:
I think this will still make an excellent addition so I'll move this forward to the upcoming release.

dennissiemensma, 2016-11-20:
It took quite some effort, but I seem to have this working now.

* Energy prices are unaffected. Users will still have to enter the same price twice, but that is mostly due to how the application technically works. It was simply too risky to alter the internal mechanism for this.

* Merging the tariffs is now a frontend/interface setting. By keeping the prices unaffected, this also means that it will work retroactively (and can be reversed again without harming existing data).

* Visible changes for the 'single tariff' affect Dashboard, Archive, Compare and Trends (lower pie charts are hidden when merging tariffs).

* I also refactored some views which seemed to get a bit slow due to growing data.

* I managed to fix all uncovered code with 100% test coverage (despite external tools still only accounting for 90+% #151).

* The dashboard no longer auto-refreshes the page. It does refresh the counters on top of the page, displaying current usage. The top now longer displays the gas or temperature. The gas/temperature graphs however are unaffected. The page stills needs a (manual) refresh to have the graphs updated.

dennissiemensma, 2016-11-20:
Merged back to `development` branch for upcoming `v1.4`. 

Docs (dev) are updated with the new feature: http://dsmr-reader.readthedocs.io/en/development/faq.html
Dutch version seems to be behind, even though it was translated. I'll look into that later.

closed 2016-11-20 by dennissiemensma

dennissiemensma, 2016-11-28:
The release containing your request has just been released: 
https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.4.0