dsmrreader/dsmr-reader #132: No data being exported

WatskeBart, 2016-05-02:
A hourly or daily data export generates no data. Time span is a single day on the calendar.
See screenshot.
[image]

dennissiemensma, 2016-05-02:
Thanks once again for the report. Does your archive page and trends page display any data yet? (because it's based on that data)

WatskeBart, 2016-05-03:
You're right once again :) Generated just fine when it had enough data.
Maybe add low and high to the T1 and T2 sections.

Replace this
[image]
with something like;
[image]

dennissiemensma, 2016-05-04:
Thanks for the suggestion. I will add verbose names to the fields, making them much more readable.

dennissiemensma, 2016-05-04:
Fixed in 0774dda:

- Added verbose names to model fields with gettext support.
- Exposed them in the csv exports.
- Added gettext translations for Dutch.

Merged back to dev, will be bundled with release 1.2.

closed 2016-05-04 by dennissiemensma