dsmrreader/dsmr-reader #134: Addition of extra costs

WatskeBart, 2016-05-03:
For a total overview of costs, two more factors could be implemented;

- Network costs - [info](http://www.energieleveranciers.nl/energie/energierekening/netwerkkosten)
- Government tax costs - [info](http://www.energieleveranciers.nl/energie/energierekening/energiebelasting)

Maybe implement a whole new menu item **Costs**

dennissiemensma, 2016-05-04:
Thank you for the idea. It has come to my mind earlier in the project, but I guess it can be implemented if requested.

Main reasons for skipping it back then:
- The project is primarily aimed at logging your consumption and plotting the data. The current costs implemented are only to have people 'sense' how much a day 'costs' on average or compared to other days.

- Maintaining the network costs and/or goverment taxes would required either me, or the end user, to keep the prices in sync. It might also result in price mismatches reflected when comparing them to the 'real' energy supplier. 

- It might also create conflicting date ranges, as they are not (always) bound to your energy contract but to each 'natural' year, starting on January the first mostly. This can be implemented technically, but might also confuse some users.

So therefor I mark this for a possible release later.

WatskeBart, 2016-05-04:
Maintaining the network cost and/or government taxes should be a user responsibility.
About the date ranges, i'm not sure what would be a good solution to conflicting date ranges.

dennissiemensma, 2016-11-17:
As this was planned originally (since May) for the next upcoming release I've reconsidered adding this feature now. 

I do, however, see too many cons of implementing this. It is still mostly because of the reasons above. Also:

* The project was (and still is) made to have insight in your daily usage and compare days / months / years with each other (#66 & #94). 

* Adding any variable taxes in the tariff could also be realized by simply altering the tariffs.

* Calculating (or having insight) in network costs can be viewed easily here: https://www.vastelastenbond.nl/energie/netwerkkosten-2016-overzicht-alle-netwerkbeheerkosten/

* It would also make any price calculations too difficult as the contract prices and date ranges differ from the 'fixed' costs this ticket is about.

* Currently more and more energy suppliers allow their customers have insight in their estimated costs as well. This was a whole different when I started this project a year ago.

I hope this will make you understand why I will skip this feature. I do however now see fit for another request of yours: #130

closed 2016-11-17 by dennissiemensma