dsmrreader/dsmr-reader #137: Colors for pie charts on trends page

WatskeBart, 2016-05-11:
Maybe give T1 and T2 different colors for better visibility.

dennissiemensma, 2016-05-12:
Because colors and the use of them is something personal, I suggest it should be an user setting.

Note for later: Check out whether something like `jaredly/django-colorfield` will be sufficient. A simple Django admin color field.

dennissiemensma, 2016-05-24:
Completed an initial setup, requires next:

- screenshots update
- compare page graph colors
- test coverage (if any dropped)

dennissiemensma, 2016-05-25:
Merged to `development` for the next release.

Preview in screenshots @ http://dsmr-reader.readthedocs.io/en/development/screenshots.html

(http://dsmr-reader.readthedocs.io/en/development/_static/screenshots/admin_graph_colors.png)

closed 2016-05-25 by dennissiemensma

dennissiemensma, 2016-07-13:
This issue is part of the next release, `1.3.0`, which was planned for next Sunday. It will still be released by then (with less issues included), also containing your request.

I don't think I've added a link to the color settings, but it's part of the frontend-settings in the admin panel, which can be found here `YOUR-URL/admin/dsmr_frontend/frontendsettings/`.