dsmrreader/dsmr-reader #138: Feature request Watt peaks

WatskeBart, 2016-05-11:
Requesting a extra counter to see the highest and lowest Watt peaks ever reached.
Maybe on the top bar of the dashboard page.

dennissiemensma, 2016-05-12:
Thank you, as this would be a nice addition to #115, which is scheduled for v1.2. 

The dashboard is meant for 'recent' data only, but I think the 'statistics' page would be a nice place for it. I added #115 to the trends page, but I'll move that also to the statistics page. This should result in having an overview of the lowest, highest and most 'constant' usage there.

dennissiemensma, 2016-05-12:
In case you are curious right now, just execute these queries over your own data.

**Postgresql**
*(Ignore any 'could not change directory to "/root": Permission denied')*
* Min:
 `sudo sudo -u postgres psql dsmrreader -c 'select min(electricity_currently_delivered) from dsmr_datalogger_dsmrreading limit 1;'`

* Max: 
`sudo sudo -u postgres psql dsmrreader -c 'select max(electricity_currently_delivered) from dsmr_datalogger_dsmrreading limit 1;'`

or

**Mysql**
* Min: 
`echo "select min(electricity_currently_delivered) from dsmr_datalogger_dsmrreading limit 1;" | sudo mysql --defaults-file=/etc/mysql/debian.cnf -v dsmrreader`

* Max:
 `echo "select max(electricity_currently_delivered) from dsmr_datalogger_dsmrreading limit 1;" | sudo mysql --defaults-file=/etc/mysql/debian.cnf -v dsmrreader`

Please note that the results are in kW and not W, so you'll have to multiply them by 1000 to get Watt.

dennissiemensma, 2016-05-12:
Merged to dev for v1.2 in 21bc7d65cd136da03e84f1ffdd485005294405ce

Updated screenshot as well, for a dev preview: http://dsmr-reader.readthedocs.io/en/development/_static/screenshots/statistics.png

closed 2016-05-12 by dennissiemensma

dennissiemensma, 2016-05-18:
`v1.2` has just been released: https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.2.0

You can upgrade by executing the `./deploy.sh` script in the root of the project (as `dsmr` user).