dsmrreader/dsmr-reader #140: API endpoint for datalogger

dennissiemensma, 2016-05-19:
There should be an alternative entrypoint for inserting data logged remotely. This could also be used for #96.

- New dsmr_api app (`api/v1/`)
- Setting for whether the API is enabled.
- Setting for the API key used (auto generate?).
- Simple view to receive a POST request containing the raw data from the meter. Should just call `telegram_to_reading()`.
- New item in docs describing this feature and examples.
- Test coverage.

Thanks to @danielterhorst for this idea.

dennissiemensma, 2016-05-19:
@danielterhorst I've completed an initial attempt for this:

* See a7f38646221a1bb048494b22b5b09515d6c1e32f and switch to branch `140-datalogger-api`.
* I've allowed RTD to build the docs for the current branch, this should provide enough information for you: http://dsmr-reader.readthedocs.io/en/140-datalogger-api/api.html

Could you try it out and provide feedback? Also any input regarding the docs above is helpful. :]

danielterhorst, 2016-05-23:
@dennissiemensma First indication is that it seems to be working :-) Will have to keep it running a little longer to see if any problems show up, but for now it's processing telegrams that were submitted using the API properly. Did not have to do anything that wasn't documented. Copied the requests example, and it works fine.

dennissiemensma, 2016-05-24:
Thanks for the input! It has been merged to the `development` branch.

closed 2016-05-24 by dennissiemensma