dsmrreader/dsmr-reader #144: CSS large margin-bottom

WatskeBart, 2016-07-20:
All pages seem to have a unnecessary large bottom margin in their tables of 20 pixels.

CSS file: /static/dsmr_frontend/css/bootstrap.min.css
Class: <table class="table table-condensed">

.table {
    width: 100%;
    margin-bottom: 20px;  << really big
}

I personally think it looks better and more compact with no bottom margin.

dennissiemensma, 2016-08-03:
Thanks for the headsup, I'll check it out when I start working on a release again, later this summer. Until then, you might 'alter' it locally by using a plugin such as Stylish, if it really bothers you :]

dennissiemensma, 2016-08-15:
I've checked it and I agree that the large margin has no use. It will be removed in the next release.

dennissiemensma, 2016-08-15:
Merged to `development` branch.

closed 2016-08-15 by dennissiemensma

dennissiemensma, 2016-08-16:
`v1.3.1` was just released, including a fix for your issue. Once again thanks for reporting.