dsmrreader/dsmr-reader #146: DSMR site behind proxy ?

hanslammerts, 2016-08-14:
Hi Dennis,

First of all, great work !
I have dsmrreader running happily on a Rpi2B, and all works perfectly.
However, I would like to use another small PC that is connected to the internet, and where all my other sites run from. Apache2 is running on this PC, and I'd like to proxy the Rpi where dsmrreader is running on on this machine. Created a new config for this to work. It look like this
`<Location /energy>
ProxyPass http://192.168.0.191/
        ProxyPassReverse http://192.168.0.191/
</Location>`
This does not work properly, however. I do get a page back, but it seems like all CSS is gone.
Can you please tell me if it is at all possible to proxy dsmrreader so it can be served by another webserver on a remote machine ?
Another solution could possibly be to run dsmrreader on the internet pc, and have a second instance run on the rpi connected to the P1 port, and have this instance fill the remote mysql database on the internet pc. Would that work ?

Regards,
Hans

dennissiemensma, 2016-08-14:
Hi thank you for your kind words. As for your issue, there is no need to run the instance somewhere else. The problem at hand is that the application signals the browser that all static files (CSS etc) should be located at `/static/`, which only works fine when viewing the application directly on the rpi.

You can easily change the URL-location of the static files (as communicated to the browser) in the config:
* Login on the terminal as the dsmr-user and change dir to the root of the project.
* Open `dsmrreader/settings.py` with a text editor (vi, nano of whatever)
* Add the following statement to the bottom of the file:
`STATIC_URL = '/energy/static/'`
* Restart the `dsmr_webinterface` process using either a graceful reload using the `./reload.sh`script in the root of the project (as dsmr user) or `supervisorctl restart dsmr_webinterface` (as root user).
* The browser should now find the static files on the right location (as the request will be proxied first to the rpi on `/energy/` and then request the files on `/static/` locally).

Please note that, as a side effect, you will no longer be able to directly view the application on your rpi, as  your browser will be unable to find `/energy/` served there. To keep it working on both locations we'll need to think of a better solution.

Let me know if this helps you.

hanslammerts, 2016-08-14:
Hi Dennis,

 

Thanks for the swift answer.

If I have the time, I’ll try to implement this tomorrow and will let you know if it is successful.

 

One other thing I noticed is, that i.e. going to the statistics page, made the browser go to (in this case) http://mywebsite/statistics, which of course 

can’t be found, because the dsmr site is proxied from /energy/ so the correct URL should be http://mywebsite/energy/statistics .

I’m curious if your suggested changes will remedy this al well.

 

Tomorrow I’ll know……

 

Thanks again,

Hans

dennissiemensma, 2016-08-15:
You are totally right, I didn't think about the other pages. So forget about changing the `STATIC_URL` setting, as it will only fix the dashboard. 
I will think of a solution for this. It shouldn't be that hard, but proxies can be a pain sometimes. :]

hanslammerts, 2016-08-15:
Thank Dennis,

 

In the mean time I’ll see if I can hack the current installation, maybe even just by hardcoding the URLs.

Just so it will work in my particular situation.

Am curious to see what solution you can find. I’m not a programmer, nor do I have any indepth knowledge

of proxies, so I’m afraid I can’t help out….;=((   (apart from being a test subject, of course)

 

Hans

hanslammerts, 2016-08-15:
Hi Dennis,

 

I have a partial success.

Went googling around a bit, and found the following :

 

http://blog.mathieu-leplatre.info/deploy-django-behind-a-reverse-proxy.html

 

So I changed ~dsmr/dsmr-reader/dsmrreader/settings.py, and added the line

FORCE_SCRIPT_NAME = '/energy’

Did have to use the line STATIC_URL too, though.

 

Now, all pages work through the proxy, except for the admin pages. For some reason this

URL is now is now expanded to https://www.hanslammerts.nl/energyenergy/admin/login/?next=/energy/export

 

I’ll see if I can figure out how to do the admin pages correctly.

hanslammerts, 2016-08-15:
Ok, once more....

 

I *think* I may have found the correct settings so you can run dsmrreader (Django) behind an Apache2 proxy.

 

In the ~dsmr/dsmr-reader/dsmrreader directory I had to add the following three lines to settings.py:

 

STATIC_URL = '/energy/static/'

FORCE_SCRIPT_NAME = '/energy'

USE_X_FORWARDED_HOST = True

 

That is, of course, when your Apache proxy setting (on the remote host) looks like this :

 

<Location /energy>

        RequestHeader set X-Forwarded-Proto 'https' env=HTTPS

        ProxyPass http://192.168.0.191/

        ProxyPassReverse http://192.168.0.191/

</Location>

 

I found the RequestHeader line somewhere else. Not sure if you need it.

Also, I tried all combinations of lines in settings.py. The only combination that worked was

the one where you have to use all three lines.

 

I would like to run this solution by you to see if this is a valid solution.

If you like you can check the result here: 

https://www.hanslammerts.nl/energy/

 

Cheers,

Hans

dennissiemensma, 2016-08-15:
Thank you for sharing your configuration and application. I'm glad te hear and see that it seems to works.

It sure is a valid solution, as long as you only alter the **webserver-configs** and the `dsmrreader/settings.py` file. Then you will not run into any conflict when updating the application later.

Please take note that:
* You should restrict your application with a password when exposing it publicly. You might have had that in mind already, but if not so, then please read the following: http://dsmr-reader.readthedocs.io/en/latest/application.html#public-webinterface-warning

* I don't have solar energy myself, so it's really cool to see that working well (I can only 'fake' it when developing). Your dashboard however displays a lot of 'zero' electricity usage. Just to be sure, that is because of the solar energy?

closed 2016-08-16 by dennissiemensma

hanslammerts, 2016-08-17:
Dennis,

 

Sorry for the late reply.

 

Yes, I was planning to put some sites behind authentication soon.

And about the zero usage…..I don’t know. I had assumed this is indeed because I’m generating electricity, and therefore the

smart meter subtracts what I use from what I generate, resulting in zero usage because I generate more than I use at those moments.

Will look at the P1 datagrams to see if this is really the case. If not, I’ll let you know, because then something’s broken in the code I guess.

 

Hans

swerveshot, 2018-08-27:
Guys, I know this is an old issue. But I'd just like to confirm that these settings are still valid and also lead to a working NGNIX reverse proxy with a subpath for dsmr-reader.

jimmyd-be, 2024-08-11:
I'm currently trying to place DSMR reader behind a Nginx proxy. I have DSMR reader installed using the docker image and do I need to reconfigure everything to make it work? Currently all the static files cannot be loaded because all the urls are wrong (base url is for me https://xxx.duckdns.org/dsmreader) and it looks like the application is trying to get the static files from https://xxx.duckdns.org/static instead of https://xxx.duckdns.org/dsmreader/static

dennissiemensma, 2024-08-14:
You can try setting STATIC_URLhttps://dsmr-reader.readthedocs.io/en/v5/reference/env-settings.html#django-static-url