dsmrreader/dsmr-reader #152: Push notifications

jeroenpeters1986, 2016-11-20:
DSMR Reader great for pulling information, even with customizable reports and all. I think it might also be a good addition to have just little summarized usage-notifications being pushed to your phone.

Since Android and iOS are the most popular mobile platforms, I'd like to propose [NotifyMyAndroid](https://www.notifymyandroid.com) (for Android) and [Prowl](https://www.prowlapp.com) (for iOS).

dennissiemensma, 2016-11-20:
I've created a branch `152-push-notifications` for your changes. Can you try creating a pull request to that specific branch?

See https://help.github.com/articles/creating-a-pull-request/

dennissiemensma, 2016-11-21:
After all it seems to be merged, Travis is now checking it in:
https://travis-ci.org/dennissiemensma/dsmr-reader/builds/177738832

dennissiemensma, 2016-11-24:
I have fixed the typo you found in 9bd9bfbcf44404f91004bcb2e37ad3c95920f493

jeroenpeters1986, 2016-11-24:
Ha, you were ahead of me, I was planning to do that tonight. Good to see that it's fixed :)

dennissiemensma, 2016-11-24:
I've tested this now and it likes like a charm :] 

I added some text formatting because my first notification was showing and electricity usage of `5.3439999999999999`. Now this (and the total prices + gas) will be rounded to two decimals.

dennissiemensma, 2016-11-24:
Merged to `development` as https://travis-ci.org/dennissiemensma/dsmr-reader/builds/178703214 succeeded.

closed 2016-11-24 by dennissiemensma

jeroenpeters1986, 2016-11-24:
Cool :+1:

dennissiemensma, 2016-11-24:
I just thought of the situation of some users not having their gas read at all. I've added a test for this in 8ecfcc253d0d557c5b09c1c22caf93b1f73d503e (using existing tests, but altering the loaded fixtures) to simulate this.

And as my own 'formatting' addition crashed, I've fixed that as well.

dennissiemensma, 2016-11-28:
The release containing your pull requests has just been released: 
https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.4.0