dsmrreader/dsmr-reader #159: Add new note has wrong time format

sevickson, 2016-11-24:
First of all, great application! Keep up the good work. First time creating an issue on Github so bear with me.

When creating a note from the Dashboard, Dashboard is in english, it puts the wrong date-format in the note. See screenshot. 
The format it accepts is yyyy-mm-dd.
[image]

dennissiemensma, 2016-11-24:
Hi! Thank you for your kind words and reporting this bug. 

I can also reproduce this bug myself, it so I'll make sure to have it fixed. :]

dennissiemensma, 2016-11-24:
I thought I made this note date variable locale independent, but I seem to only have applied that to the same button in the Archive page. 

Now it's fixed in 9af380679e3893cc0b2e46e63b2aecef98b8e23e and will be bundled with the upcoming `v1.4` release.

dennissiemensma, 2016-11-24:
@sevickson all contributers are listed on the credits page: http://dsmr-reader.readthedocs.io/en/latest/credits.html. 

I will list you als "sevickson" by default, but if you'd like to have your real name listed there, just let me know here.

dennissiemensma, 2016-11-24:
Tests greenlit @ https://travis-ci.org/dennissiemensma/dsmr-reader/builds/178648725

So hereby merged back to the main `development` branch for v1.4.

closed 2016-11-24 by dennissiemensma

sevickson, 2016-11-24:
@Dennis:

It is my real name. :-P

dennissiemensma, 2016-11-24:
Great! Then **Sevickson** it is, thanks again. ;-)

dennis, 2016-11-25:
@sevickson Please don't use @dennis unless you want to notifiy me :)

dennissiemensma, 2016-11-28:
The release containing the fix for your report has just been released: 
https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.4.0