dsmrreader/dsmr-reader #160: Update documentation for notifications

jeroenpeters1986, 2016-11-26:
Notifications are now "live" in the developmentbranch. A little explaination about both notification services should lead users the way in setting them up

jeroenpeters1986, 2016-11-26:
I'll make a PR for this :)

jeroenpeters1986, 2016-11-26:
@dennissiemensma did you use specific tools to translate the documentation?

dennissiemensma, 2016-11-27:
Many thanks. The v1.4 release was already frozen, but since it's only doc updates I've merged it.

I also updated the docs HOWTO in 0186f787450cd56a55d1c501cb0d5bbbd3ec5be7, as it seems possible after all to preview docs translation results.

closed 2016-11-27 by dennissiemensma

dennissiemensma, 2016-11-27:
RTD also built the docs now, looking good:

* http://dsmr-reader.readthedocs.io/en/development/faq.html

* http://dsmr-reader.readthedocs.io/nl/development/faq.html

jeroenpeters1986, 2016-11-27:
Nice, looking good indeed 😎