dsmrreader/dsmr-reader #161: Fases/costs

pvink, 2016-11-26:
I have a suggestion to add fases.  So you can see the power usage per fase. Also it would be nice if you can hang a seperate cost category on when you give power back to the grid.

Awesome software by the way!

jeroenpeters1986, 2016-11-26:
Hi pvink, just checking if you mean 'groups' by fases? The DSMR reader reads messages from the power measurement instruments, this means is is measure BEFORE the groups. So I'm not sure if that would work

pvink, 2016-11-26:
My smart meter's telegram return's the current total power consumption but also seperated in 3 groups (for my case).

Is it also possible to see the power usage of the current day? Now I am only  able to see the last hour from the current day.

jeroenpeters1986, 2016-11-26:
hi pvink, I think you will help Dennis a lot if you can post a few examples of the telegrams here :)

dennissiemensma, 2016-11-27:
@pvink thank you for your requests. Since you are talking about three different things here, I've moved two of them to #163 and #164.

Regarding your question about phases, could you provide a bit more information? As Jeroen already suggested, a dump of your telegram might help a lot. Please be sure to remove any unique meter ID's from it before posting it here.

pvink, 2016-11-28:
Thank you for splitting them. My telegram is here:

1-3:0.2.8(42)
0-0:1.0.0(161122180614W)                    //time notation J/M/D/U/M/S
0-0:96.1.1(lot's of numbers)
1-0:1.8.1(000150.503*kWh)                   //total power usage 'dal'
1-0:1.8.2(000220.871*kWh)                   //total power usage 'piek'
1-0:2.8.1(000001.694*kWh)                   //total return power usage 'piek'
1-0:2.8.2(000003.983*kWh)                   //total return power usage 'piek'
0-0:96.14.0(0002)                           //current power rate (piek/dal)
1-0:1.7.0(02.480*kW)                        //total current power usage
1-0:2.7.0(00.000*kW)                        //total current return power
0-0:96.7.21(00001)          
0-0:96.7.9(00001)
1-0:99.97.0(1)(0-0:96.7.19)(000101000020W)(2147483647*s)
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(000*A)
1-0:51.7.0(011*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.016*kW)                               //current power usage - group 1
1-0:41.7.0(02.455*kW)                               //current power usage - group 2
1-0:61.7.0(00.009*kW)                               //current power usage - group 3
1-0:22.7.0(00.000*kW)                               //current return power - group 1
1-0:42.7.0(00.000*kW)                               //current return power- group 2
1-0:62.7.0(00.000*kW)                               //current return power- group 3
0-1:24.1.0(003)                                     //gas valve position
0-1:96.1.0(XXXX)      //id gasmeter 
0-1:24.2.1(161122180000W)(00184.342*m3)             //last transmitted gas usage to company
!E4FE

My english very very bad at the moment, I hope you get it what I mean.

dennissiemensma, 2016-11-28:
Thank you for your input. Now I see what you are pointing at. 

That piece of information is currently discarded by the application because it wasn't used as input for storing and graphing the data.

For now I'll move your request to the backlog, as I think your other two requests will help a whole lot of other users. I'm not sure whether and when I can implement the phases, as it also required a lot of data migrations and additional graphs. On hold for now.

Jeltel, 2016-12-12:
I'm also very interested in power usage of the phases. 
Always good to be able to see if one of the phases is being over used.

dennissiemensma, 2016-12-12:
Hi, thank you for 'voting' for this feature, I've added it now to the `v1.6` feature list, which is currently scheduled for February.

JeroenTielen, 2016-12-13:
Thanks Dennis for adding this in the 1.6v. I'm also very interrested to see this as I'm also having this 3 phases setup at home. 

Great software btw, much appreciated.

dennissiemensma, 2016-12-13:
Hi @pvink / @Jeltel / @JeroenTielen, maybe you guys could elaborate a bit on what kind of info you are expecting. 

Would it suffice to just see the recent phase(s) usage on the dashboard plotted in a graph? Possibly combined with #164 to increase the time range displayed there a bit.

Jeltel, 2016-12-13:
Thats pretty much it for me. Maybe save the max value and time per phase.

For me it's to check if one of the phases is maxed out. Maybe I'll switch some groups to another phase if needed.

JeroenTielen, 2016-12-13:
That would work for me as well. Maybe a checkbox so you can switch between combined or per phase during viewing (if that is possible)

pvink, 2016-12-20:
Basically what Jeltel and JeroenTielen are saying. If you can catch the value or the option is enabled ( just like you did on the gas, weather and return energy) you can show it in a different graph. So with 3 phases a graph for each and a graph for total consumption on the top?

Even the 'return energie' can be splitted when you look at my telegram. Now you are doing 'energy consumpton' minus 'generated energy'..which resulsts in a - or + value and thus you can't see your actual generated power/power consumption. 

## **So my telegram shows 6 different power consumption numbers**
1. **Total power consumption** since installation of the meter of the **day usage**
2. **Total power consumption** since installation of the meter of the **night usage**
3. **Current power consumption** at this moment. It has another line which shows if it is day or night power consumption.
4/5/6. Current power consumption per phase

The above also counts for generated power. So in total there are 12 different value's. It is a lot, haha.

My personal advice is that you leave the 3 phases as it is and just focus on splitting the power consumption and the generated power.

Can't wait for the next update 👍

dennissiemensma, 2016-12-21:
Hi guys, I finished a proof of concept yesterday regarding plotting the phases in a graph.

- It's only displayed when there are three phases available (I think I will still make it a setting as well and hide it by default).

- The phases are all plotted in a single graph.

- By coincidence it happens that ChartJs 2.x (#127) by default allows you to enable or disable any data set by clicking on it. So you would be able to hide one or two phases while viewing it. 

- I've linked the phases graph to the same scroll bar as for the electricity (which is freshly introduced as well, for #164). So you can scroll back as well, for a limited offset (see #164).

Below is a screenshot where I triggered multiple phases shortly:

[image]

It this were you are looking for?

dennissiemensma, 2016-12-21:
@pvink I'm not sure whether your *"So my telegram shows 6 different power consumption numbers"* fits in here. And I'm having trouble to understand your expectations regarding that.

Just so you know:
- It's impossible to know how much power is generated exactly, as some power is distributed into your home directly, without even reaching the smart meter (see #145 as well). You will need to plot the data of your solar transformer as well (which actually does know the generated value). And that is outside the scope of this project.

- The 'electricity returned' displayed by this application is the actual value which is returned to the net (outside your home). It does not have nor subtract any power generated, as it simply does not know. The same applies to the `P-` of the phases (as only `P+`, consumption, is plotted in the screenshot above). I'm not sure if `P-` will provide that information as well. It surely can not be used to tell how many Watt hours are generated, only the `(k)Watt` flowing at the exact moment of when the reading was taken.

So please clarify. :]

The only thing I can do here as well, is to store and plot the 'return' phases (`P-`) similar to the example above.

Jeltel, 2016-12-22:
That's great Dennis. You can see the wattage on every phase and the total. Al great. And the feature to scroll back a week is very welcome as well.

Next is the max value for one phase. So you can check if you need to redistribute the electrical devices.

dennissiemensma, 2016-12-22:
@Jeltel thank you for your feedback. One more question:

- *Do you think plotting the `P-` (return phases used for solar energy) will add any value?*

As for your other request, I do not see a fit for displaying the max value of a phase, because it will keep displaying the (old) highest value after redistributing. And make stuff more complex as well.

**However**, that does not mean that I cannot provide you the information you ask for. You can easily query the database for the information you are looking for, with a one liner.

- For example, plotting the highest value + timestamp of a single phase would be something like:

`sudo -u postgres psql dsmrreader -c "select read_at, phase_currently_delivered_l1 from dsmr_consumption_electricityconsumption order by phase_currently_delivered_l1 desc limit 5;"`

And display (fake 'old' data here):
```
        read_at         | phase_currently_delivered_l1 
------------------------+------------------------------
 2016-11-19 18:39:00+01 |                        4.105
 2016-10-27 17:00:00+02 |                        3.803
 2015-12-20 17:07:00+01 |                        3.799
 2016-08-19 21:15:00+02 |                        3.710
 2015-12-20 17:05:00+01 |                        3.701
```

- And after redistributing your groups, you can easily query everything **after** that specific moment (let's say December 1st for example):

`sudo -u postgres  psql dsmrreader -c "select read_at, phase_currently_delivered_l1 from dsmr_consumption_electricityconsumption where read_at::text > '2016-12-01 00:00:00' order by phase_currently_delivered_l1 desc limit 5;"`

Which will ignore anything **before** that moment picked:
```
        read_at         | phase_currently_delivered_l1 
------------------------+------------------------------
 2016-12-12 17:33:00+01 |                        3.513
 2016-12-14 22:09:00+01 |                        3.053
 2016-12-14 22:08:00+01 |                        3.035
 2016-12-14 22:10:00+01 |                        3.019
 2016-12-05 17:34:00+01 |                        2.872
```
So by the time this is released, and you have recorded the data you are looking for, I can surely help you with that.

dennissiemensma, 2016-12-27:
- @pvink / @JeroenTielen do you guys have anything to add [regarding this proof of concept](https://github.com/dennissiemensma/dsmr-reader/issues/161#issuecomment-268607833)?

- @Jeltel do you think plotting the P- (return phases used for solar energy) will add any value? Personally I don't think it will, but I'm not having any solar panels at home anyway...

This ticket is the last one regarding a (new) feature for the upcoming release. It would be a shame if it get's postponed after all. 

Just let me know whether you guys think this needs more reworking before merging it into the next release...

JeroenTielen, 2016-12-27:
No comment from me. This looks perfect. Love the seperation in graphs from total and the phases. 

Thumbs up.

Jeltel, 2016-12-27:
I don't have panels either.
Only reason to use it is where you have a lot of panels and are able to get more than the allowed amperage out of the panels.
Of so, you have a hell of an installation and already should have thought of it.

So no, I don't think it really adds something.

dennissiemensma, 2016-12-28:
Thanks for your input guys. Then I will merge this for the upcoming release, as is.

dennissiemensma, 2016-12-28:
The [test seem to pass (after a translation fix)](https://travis-ci.org/dennissiemensma/dsmr-reader/builds/187275930), so merged back to `development` now.

Also deployed successfully here.

closed 2016-12-28 by dennissiemensma

Jeltel, 2016-12-28:
Very nice. After turning it on, you'll have to wait a bit for readings to turn up. Otherwise great! Gives even more insight on your usage :)

dennissiemensma, 2016-12-28:
Great, keep [this comment in mind](https://github.com/dennissiemensma/dsmr-reader/issues/161#issuecomment-268869066) when you want to check the peaks of each phase.

dennissiemensma, 2017-01-01:
@pvink / @JeroenTielen the [`v1.5` release is out](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0), containing this issue.

Please mind:
- [the release notes](http://dsmr-reader.readthedocs.io/en/latest/changelog.html#v1-5-0-2017-01-01) 
- [how-to-upgrade docs](http://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-update-my-application)

Also note that upgrading requires you to cd into the project root and activate the VirtualEnv, after switching to the `dsmr` user. The installation guide hinted that before, but it's [highlighted more clearly now](http://dsmr-reader.readthedocs.io/en/latest/installation.html#virtualenv).

JeroenTielen, 2017-01-02:
Hi Dennis, got it up and running. But I need to click on "Display Electric Phases" every time to display the graph. I didn't found any configuration setting to always display this. 

Can it be another feature request to always display the graph? ;) (So no extra click is involved)

dennissiemensma, 2017-01-02:
@JeroenTielen great to hear.

Would you mind creating a new issue for your request? The current issue is closed/completed/merged.