dsmrreader/dsmr-reader #163: Allow separate prices/costs for electricity returned

dennissiemensma, 2016-11-27:
Currently only the electricity consumption (and gas) support prices and costs display. It would be nice to have (separate) prices support for electricity **returned** as well.

[Initially reported by @pvink]

mkruiver, 2017-01-26:
Currently, the cost display is basically not correct as electricity returned is not considered at all... I understand that implementing this could be tricky with respect to 'saldering' and nett return, but would be nice to have this implemented!

dennissiemensma, 2017-09-28:
Ik heb hier een toevoeging voor gemaakt. Het is straks mogelijk om een tarief in te stellen voor teruglevering (zowel hoog- als laagtarief). De totaalkosten van een dag worden daar tevens in meegenomen (dus kan negatief uitvallen wanneer voldoende teruggeleverd wordt).

closed 2017-09-28 by dennissiemensma