dsmrreader/dsmr-reader #164: Display last 24 hours on dashboard

dennissiemensma, 2016-11-27:
The current dashboard only displays the last 30 electricity readings. It would be great if it'd be possible to display the last 24 hours there to have more insight of the current day.

[Initially reported by @pvink]

dennissiemensma, 2016-12-14:
I don't think the current graphing tool will be able to render an entire day's x-axis in the dashboard. As I will also work on #127 I'll check whether there are any alternatives for this. 

Maybe fetching the data in the background using XHR and `<` + `>` buttons to 'scroll' around.

dennissiemensma, 2016-12-19:
Merged to `development`.

closed 2016-12-19 by dennissiemensma

dennissiemensma, 2016-12-20:
Screenshot: http://dsmr-reader.readthedocs.io/en/development/_static/screenshots/dashboard.png

pvink, 2016-12-20:
Looks good! This is perfect for me! Nice update

dennissiemensma, 2017-01-01:
@pvink  the [`v1.5` release is out](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0), containing this issue.

Please mind:
- [the release notes](http://dsmr-reader.readthedocs.io/en/latest/changelog.html#v1-5-0-2017-01-01) 
- [how-to-upgrade docs](http://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-update-my-application)

Also note that upgrading requires you to cd into the project root and activate the VirtualEnv, after switching to the `dsmr` user. The installation guide hinted that before, but it's [highlighted more clearly now](http://dsmr-reader.readthedocs.io/en/latest/installation.html#virtualenv).