dsmrreader/dsmr-reader #165: Capability based push notifications

dennissiemensma, 2016-11-27:
The recently added feature in #152 provides notifications about electricity and gas consumption. There are however users without gas consumption or users who also return electricity to the net.

It would be nice to construct the notification message based on the capabilities of the application environment.

jeroenpeters1986, 2016-11-27:
I think I could make this change, is the gas-usage always 0, or does this
situation presents itself some other way?

notifications@github.com>

dennissiemensma, 2016-11-27:
Thanks! But I'm already working on it. There is a service called `get_capabilities()` which can provide this information.

jeroenpeters1986, 2016-11-27:
Cool!

dennissiemensma, 2016-11-27:
Merged to `development` as RC5. Also delayed v1.4 for a few days.

closed 2016-11-27 by dennissiemensma