dsmrreader/dsmr-reader #169: Other port

pvink, 2016-11-29:
Hello,

Another question from me :)
Is it possible to use another port for the web interface? 

I am trying to run also pi-hole on my pi, but it seems that program can't change it default port.  I can't seem to find a configuration file or anyother file where I can locate the port.

Thanks!

dennissiemensma, 2016-11-29:
Hi,

Yes you can change it to any port for dsmrreader, but it's not an application setting.

If you followed the installation guide from the docs, you are running a Nginx web server in front of the application. It's also the one your browser 'talks' with.

The web server determines which ports are used for serving applications. And you can change it quite easily:

- Make sure you are logged in on your rpi as `root` user or a user with `sudo`-capabilities.
- Open Nginx configuration file: `sudo vi /etc/nginx/sites-enabled/dsmr-webinterface`, you should see something similar as:
```

server {
    listen      80;
    server_name _;  # No hostname available? Just use the underscore for wildcard matching. :]

```

- Change the directive `listen      80;` to anything you want, such as `listen      8080;`.
- Save the file.
- Have Nginx check you didn't make any typo's: `sudo service nginx configtest`.
- It should say OK.
- Now apply changes safely: `sudo service nginx reload`.
- The application should now be accessible on the port you changed it to.

closed 2016-11-29 by dennissiemensma

pvink, 2016-11-30:
Thanks! Great support you have. 👍

HFeenstra, 2017-05-26:
Hi, well I tried your steps, I got to the step, change the directive: but here is the thing, I can't type in it?
Tried different keys and all of a sudden I had changed lines, like added letters and so on, so quite the session and restarted it, but it prompted me first with a swap file.

So for me typing or saving, doesn't work, lack of knowledge I guess. Can someone point me out what to do to get this working and changing to a desired port?

dennissiemensma, 2017-05-26:
I can't help you with editing text files, since the application require users to have some basic knowledge to work in Linux in the first place.

I can however point you to some guides. You may want to try install `vim` or `nano`, where `nano` is the least complex one to use I guess (for new users):
* https://www.howtogeek.com/howto/42980/the-beginners-guide-to-nano-the-linux-command-line-text-editor/
* https://www.howtogeek.com/102468/a-beginners-guide-to-editing-text-files-with-vi/

You can install then with `sudo apt-get install -y vim nano`.