dsmrreader/dsmr-reader #170: Weather graph not showing

Helloitsme, 2016-12-06:
Thank you for the great program! Measurements from the meter are working great.
One issue however:  although weather is selected in the settings, there is no weather graph visible on the dashboard.

[image]
[image]

dennissiemensma, 2016-12-06:
Hi! Thank you for reporting your issue. This is a bug I seem to have forgotten to fix while implementing it in #18.

It is 'caused' by the lack of gas readings in your situation, because I decided (in the earlier stages of this project) to have the gas reading trigger the weather readings every hour. This works for many users, including myself, but obviously not for anyone without a gas meter connected as well.

As this is a bug that should be fixed, I will schedule this for the upcoming release. Which is planned for mid January.

Helloitsme, 2016-12-06:
Great, thanks!

dennissiemensma, 2016-12-16:
I've fixed this and I've running it now for a day or so myself. Works fine, so it'll be merged back to the main `development` branch for the next release.

dennissiemensma, 2016-12-16:
@Helloitsme please notice #180 as well.

closed 2016-12-16 by dennissiemensma

Helloitsme, 2016-12-16:
Hi Dennis, You can mention me, but i don't see myself as a tester

dennissiemensma, 2016-12-16:
Aah I see, maybe I wasn't clear enough then. 

It's just a 'wall of fame' for the project. It doesn't make you a tester or demand that you test stuff. It's just a way to say 'thank you' for your contribution (which can be a simple as reporting an issue). It's similar to pages like these: https://sentry.io/thanks/

It's optional anyway, but just so you know. :]

dennissiemensma, 2017-01-01:
@Helloitsme the [`v1.5` release is out](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0), containing this issue.

Please mind:
- [the release notes](http://dsmr-reader.readthedocs.io/en/latest/changelog.html#v1-5-0-2017-01-01) 
- [how-to-upgrade docs](http://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-update-my-application)

Also note that upgrading requires you to cd into the project root and activate the VirtualEnv, after switching to the `dsmr` user. The installation guide hinted that before, but it's [highlighted more clearly now](http://dsmr-reader.readthedocs.io/en/latest/installation.html#virtualenv).

Helloitsme, 2017-01-02:
It works, Thanks!