dsmrreader/dsmr-reader #174: Consumption chart hangs due to unique_key violation

kvolleberg, 2016-12-09:
Hi Dennis,
First off: what an excellent app, and with your installation docs I was up-and-running within only 5 minutes or so. 

During the week, I noticed that my consumption chart didn't update anymore. Possibly after a restart or so.
I checked the dsmr_backend log and noticed a duplicate_key violation error in the ElectricityConsumption model. 

By inserting the data in the model (in the dsmr_consumption.py compact routine), apparenty it blocks when a duplicate timestamp is present. You assume uniqueness of each reading, therefore no check is made if this is really the case. This fails when the reading object has processed=False, while it's not imported in the ElectricityConsumption model yet.

I updated the code in the dsmr_consumption/services.py script a bit to solve this, and I want to share this with you. Maybe you have a better solution, but it helped me for now...

Below you'll find the part I updated. 
'
    # Electricity should be unique, because it's the reading with the lowest interval anyway.
    if grouping_type == ConsumptionSettings.COMPACTOR_GROUPING_BY_READING:
        try:
            ElectricityConsumption.objects.get(read_at=dsmr_reading.timestamp)
        except ElectricityConsumption.DoesNotExist:
            ElectricityConsumption.objects.create(
                read_at=dsmr_reading.timestamp,
                delivered_1=dsmr_reading.electricity_delivered_1,
                returned_1=dsmr_reading.electricity_returned_1,
                delivered_2=dsmr_reading.electricity_delivered_2,
                returned_2=dsmr_reading.electricity_returned_2,
                currently_delivered=dsmr_reading.electricity_currently_delivered,
                currently_returned=dsmr_reading.electricity_currently_returned
                )
    # Grouping by minute requires some distinction and history checking.
'

Cheers, Koen

dennissiemensma, 2016-12-09:
Hi Koen, thank you for your compliments and especially for reporting this issue!

I have never seen duplicate source readings before, but I think it makes sense to add a check similar to your suggestion, since it seems to happen.

Could you provide some more info:
* What database backend are you running?

dennissiemensma, 2016-12-10:
I've simulated the issue and  merged a fix in edeccd9924fece074be7f558bfc414954a948d13 to the `development` branch for the next release. It will presumably be released next month.

You will also be added as contributor now, see the development version @ http://dsmr-reader.readthedocs.io/en/development/credits.html. If you'd rather not like to be put there, just let me know.

closed 2016-12-10 by dennissiemensma

kvolleberg, 2016-12-10:
hi dennis,
you've done things rather quickly. in reply to your first question: I have
a postgresql (9.4) database running.
I have no problems you mentioning me.

if I run into more issues, or think of any enhancements, I'll keep you
posted.
Enjoy the weekend!
Cheers, Koen

dennissiemensma, 2016-12-11:
Okay great! Thanks for your input and any possible feedback you'll might have in the future. :]

dennissiemensma, 2016-12-12:
@kvolleberg due to another blocking issue (#175), I decided to release a bug fix patch for these issues today.

Upgrading should be easy by running a single script: http://dsmr-reader.readthedocs.io/en/latest/application.html#application-updates-bug-fixes-new-features

Please make sure to revert any of your local changes before upgrading.

Jeltel, 2016-12-25:
I'm seeing the same issue. My recent electricity graph is stuck. Gas consumption is fine.

I want to check the logs, where is it?

dennissiemensma, 2016-12-25:
Hi @Jeltel can you [try this suggestion](https://github.com/dennissiemensma/dsmr-reader/issues/178#issuecomment-267129678)?

I asume you are running the latest version (`v1.4.1` at the moment)? And does your Status-page display any errors?

Jeltel, 2016-12-26:
Status is fine. Data is being read and saved in the database. I can see yesterday's usage in the 'archief' section.

Also the logs (after replacing the conf.d) don't show something different. At least I don't see it. I'm tailing dsmr_datalogger (shows the p1 reading), dsmr_backend (shows broadcast signal and sleeps for 5 seconds) and dsmr_webinterface.

dennissiemensma, 2016-12-26:
Well then it can not be this current issue, which prevents data from being updated. 

Maybe #178 again?