dsmrreader/dsmr-reader #178: DSMR webinterfacenot completly loading

Jeltel, 2016-12-14:
I'm using DSMR to log my meter, and this is working great. And offcourse I want to see some graphs. This worked well the last week, but now it doesn't completely load anymore.

The top of the page is loaded, but not the current Watt. The graph section is never loaded completely.
[image]
I tried restarting, to no avail.

I'm also using pi-hole (raspberry dns server with integrated ad-block for the network), which is slower when dsmr is in this state. 
Also 'top' is showing dsmr by far having the most CPU-time.
[image]

Is there some log file where I can see what DSMR is doing?

dennissiemensma, 2016-12-14:
@Jeltel thank you for reporting this issue.

- You should start by checking the **Status-page** in the application, does it state any warnings or errors? It should at least show you whether the data is being processed correctly.

If that doesn't display any error:
- Most logs are located in Supervisor, but it's a bit lagged due to Python buffering. I fixed this yesterday by forcing unbuffered mode in Python, it might help you debugging as well. 

Just replace the dsmrreader *.conf files in `/etc/supervisor/conf.d/` by this single one: https://raw.githubusercontent.com/dennissiemensma/dsmr-reader/development/dsmrreader/provisioning/supervisor/dsmr-reader.conf

Then (sudo) access `supervisorctl` and enter `reread` + ENTER, followed by `update` + ENTER. It should update some processes.

If they all state `RUNNING` then you can also view their log in supervisorctl by entering: `tail -f dsmr_datalogger` or `dsmr_backend`

dennissiemensma, 2016-12-14:
And regarding the CPU usage, 3% is actually very low. You should read the top bar of `top` as well to check the global system status.

For example:
`%Cpu(s):  0,7 us,  0,2 sy,  0,7 ni, 98,3 id,  0,1 wa,  0,0 hi,  0,1 si,  0,0 st`

* `0,7 us` = actual usage %
* `0,2 sy` = used by the operating system %
* `0,7 ni` = used by processed in `nice` mode (they will be skipped when the load is high) %
* `98,3 id` = idle % (CPU just waiting around which is good)
* `0,1 wa` = CPU % waiting on the disk (common issue for slow sd-cards)

Also check the load average: `load average: 0,10, 0,09, 0,12` (last min, 5 min and 15 mins if I'm correct)
The load should be below 1, but that also depends on the number of cores your CPU has (complex stuff).

dennissiemensma, 2016-12-14:
And finally, please note that I released a bug fix patch `v1.4.1` yesterday as well. You might be affected by it, as you run `v1.4.0`, even though the symptoms differ a bit.

https://github.com/dennissiemensma/dsmr-reader/releases

Please start by checking out your Status-page because it might be caused by something completely different.

Jeltel, 2016-12-14:
At this moment, the status is fine. Earlier today the readings where not processed by dsmr. (A gap between 11:00 and 19:00 - the last restart).

Still no Watt at the top though.

All processes have the status RUNNING.

I'll update first before doing anything :).

THanks for the tips. I might not use them today, but I probably will next week.

dennissiemensma, 2016-12-14:
You might want to check the console of your browser as well. 

Could you press `F12` and reload the Dashboard page? Maybe it's caused by some Javascript error, as you state that the readings are OK in the status page.

Jeltel, 2016-12-14:
`Using //@ to indicate sourceMappingURL pragmas is deprecated. Use //# instead[Meer info] 
TypeError: ERROR.captureStackTrace is not a function[Meer info] 
Gebruik van getPreventDefault() wordt niet meer ondersteund. Gebruik in plaats daarvan defaultPrevented.`

Jeltel, 2016-12-14:
And because I understand you do not want to bughunt all the browsers in the world. I'm happy to use Chrome from now on :)

dennissiemensma, 2016-12-14:
Thank you for your input. I have never seen that error before, but I do see that there are two files in this project which have such statement.

Could you try removing the `//@ sourceMappingURL=jquery.min.map` from these two files just to test whether it will fix it?
* `/var/www/dsmrreader/static/dsmr_frontend/js/jquery-2.0.2.min.js`
* `/var/www/dsmrreader/static/dsmr_frontend/js/jquery-1.9.1.min.js`

Please make sure to refresh your browser hard or empty the cache.

This is a bug and it should be fixed. :]

Jeltel, 2016-12-14:
I'll do that after the weekend :) I promise. Thanks for the heads-up so far :)

dennissiemensma, 2016-12-14:
I see that I can reproduce it myself now as well. No need to test it anymore, Chrome will be the alternative for now, as you suggested that yourself. :]

Jeltel, 2016-12-14:
If I'm able to help after the weekend let me know.

dennissiemensma, 2016-12-14:
Hi, I need to rectify some stuff I mentioned before.

* I can reproduce the warnings (in Firefox), but I can not reproduce the dashboard graph failing. I thought I could, but the cause for that seems the lack of data on my testing environment. When I test on my 'production' environment (which has recent data) the graphs renders well.

* The warnings do not seem to be the cause, as it works here, with the warnings displayed in the console as well.

So can you (after the weekend) :
* forget about removing the headers of the Javascript files mentioned above.

* manually run the `post-deploy.sh` script in the root of the project, as `dsmr`-user? It will check static files, database migrations and also flush any application cache. Just to be sure.

* flush your Firefox cache and retry? Or disable the cache.

* write down the Firefox version you are using?

Jeltel, 2016-12-19:
Back home and the SD card is corrupt. So I'll have to start over. When I reproduce the result I'll answer the questions you asked me.

So on hold?

dennissiemensma, 2016-12-19:
That's a bummer, do you have any backup's of the data? And what kind of SD card did you use? Yeah, it'll be on hold then.

Jeltel, 2016-12-19:
I have a Dropbox backup, but that's only settings probably?

dennissiemensma, 2016-12-19:
No it should be a complete backup of all data. You can try to open it with a text editor to view it's contents (unless it's huge).

* As you said before you were using PostgreSQL, this should restore it:
`sudo sudo -u postgres psql -f <PATH-TO-BACKUP.sql> dsmrreader`

* You should be able to execute the command above after creating the database (chapter 1 of install guide,`Database backend`): http://dsmr-reader.readthedocs.io/en/latest/installation.html#database-backend

* If successful, you can skip the first step in chapter 8: `Bootstrapping`, which is executing the `./manage.py migrate`

Jeltel, 2016-12-19:
It's only 1mb, so I'll try this later.

I like to say that you are really stimulating people to participate. The way you almost instantly give very straightforward tips and tricks is unique. Keep it up!

dennissiemensma, 2016-12-19:
Thank you for your kinds words. It's always a pleasure to help.

I have to honest that I'm not always this active. During the *spring/summer/holiday months* the development is on hold most of the time, so I won't respond this quickly all the time.

Let me know later whether you were successful in restoring the data.

Jeltel, 2016-12-20:
I was successful in restoring the data. I rerun the complete installation guide. And adding your backup line right after creating the database.

So far so good on the graphs.

dennissiemensma, 2016-12-21:
Great to hear you were able to restore the data. 

Can you see whether all supervisor processes are running? It might just be a webserver collision, as someone asked me in #169 how to run either on another port.

Jeltel, 2016-12-21:
Dsmr is running on port 8080. It's something different. The pi is not connectable over SSH after some time. I'll figure it out somehow.

dennissiemensma, 2016-12-21:
Well, let me know if you need any help. And whether you can reproduce your graph error, which was your actual issue. :]

Jeltel, 2016-12-24:
It's running fine for a few days now. Only real difference is that I've excluded the synology nas of pi-hole.

So this may all just be a glitch when using pi-hole and dsmr on the same pi and using a synology nas...

I think this can be closed.

dennissiemensma, 2016-12-24:
Hmmm still weird, but good to hear that it's resolved. Thank you for the update. :]

dennissiemensma, 2016-12-26:
@Jeltel is it the same issue as before? With Firefox?

Jeltel, 2016-12-26:
Nope this is on chrome android. And the graph itself is showing. Just old data.

dennissiemensma, 2016-12-26:
Can you tell me whether you've enabled the `Keer dashboard-grafieken om` in `Interfaceconfiguratie` settings?

Jeltel, 2016-12-27:
I did not. I did, however, had enabled `Voeg tarieven samen`.

When disabling `Voeg tarieven samen`, it still doesn't work. 

When enabling `Keer dashboard-grafieken om`, it does work.

It seems that the `Voeg tarieven samen` option is broken.

dennissiemensma, 2016-12-27:
Thanks a lot for providing this info. I hope I can reproduce it now myself and release a patch for it.

dennissiemensma, 2016-12-27:
@Jeltel unfortunately I'm unable to reproduce it. I was hoping it would crash right away here as well, or at least show outdated data.

Just to be sure:
* You **do** see a graph on the dashboard with your consumption, but it's displaying **outdated data**?
* You've only enabled the `Voeg tarieven samen` option in the frontend setting? (disregard the colors)

I've created a branch which slightly shows a more verbose graph (see below). Would you mind checking it out for a bit?

Branch:
* `sudo su - dsmr`
* `git checkout -b v1.4.1---178-fix-merge-electricity-tariffs origin/v1.4.1---178-fix-merge-electricity-tariffs`
* CTRL + D (back to root/sudo user)
* `sudo supervisorctl restart dsmr_webinterface`

Now you should see a (bigger) graph on the dashboard which also shows the date + seconds of the x-axis. 
* Can you confirm the data you are seeing is outdated? Or create a screenshot or something?
* Can you also run the following command as `dsmr` user on the command line? `./manage.py dsmr_dump_status`. **And paste it's output here**, it should not contain any sensitive or personal data.

When done, go back to latest release (as you were):
* `sudo su - dsmr`
* `git checkout master`
* CTRL + D (back to root/sudo user)
* `sudo supervisorctl restart dsmr_webinterface`

[image]

Jeltel, 2016-12-27:
Yes and yes.

And I would love to, but:
`dsmr@pi-hole:~ $ git checkout -b v1.4.1---178-fix-merge-electricity-tariffs origin/v1.4.1---178-fix-merge-electricity-tariffs
fatal: Not a git repository (or any of the parent directories): .git`

dennissiemensma, 2016-12-27:
Oops, I forgot the `git pull` command. I've updated my comment.

Jeltel, 2016-12-27:
I don't understand. I get the same error. Should be straightforward right?

dennissiemensma, 2016-12-27:
Hmmm weird, I never was a fan of `git` anyway. Still not.

What does `git branch` say?

Jeltel, 2016-12-27:
Same message...

dennissiemensma, 2016-12-27:
And I assume you are executing this in the root of the `dsmr-reader` checkout/project as well? The directory where `manage.py` is located?

Jeltel, 2016-12-27:
Whoops. Jackpot.

I see data from the 22nd.
[image]

The dump isn't working however:
`dsmr@pi-hole:~/dsmr-reader $ ./manage.py dsmr_dump_status
Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ImportError: No module named 'django'`

dennissiemensma, 2016-12-27:
I see. I'd advise you to both put these command in the `.bashrc` of the `dsmr` user:

```
source ~/.virtualenvs/dsmrreader/bin/activate

cd ~/dsmr-reader
```

The first will switch Python paths to the environment used by the application (which contains `django`, which errors because it's unknown)

The second one will prevent the `git` errors as well, putting you in the right directory by default.

The bashrc file can be found here:
* `su - dsmr`
* Edit `~/.bashrc`

See http://dsmr-reader.readthedocs.io/en/development/installation.html#virtualenv

dennissiemensma, 2016-12-27:
Don't forget to CTRL+D + `su - dsmr` again. 

You should both be in the right directory and see a prefix on your terminal: `(dsmrreader)dsmr@hostXXX`

Jeltel, 2016-12-27:
Yes! 

`pi@pi-hole:~ $ sudo su - dsmr
(dsmrreader)dsmr@pi-hole:~/dsmr-reader $ ./manage.py dsmr_dump_status

--- Capabilities ---
weather                                           : True
electricity                                       : True
any                                               : True
gas                                               : True
electricity_returned                              : False

--- Settings ---
DataloggerSettings.track                          : True
DataloggerSettings.track_meter_statistics         : True
DataloggerSettings.dsmr_version                   : 4
ConsumptionSettings.compactor_grouping_type       : 2
FrontendSettings.reverse_dashboard_graphs         : True
FrontendSettings.merge_electricity_tariffs        : True

--- Counts ---
DsmrReading.unprocessed.count()                   : 4
DsmrReading.processed.count()                     : 126219
ElectricityConsumption.count()                    : 31695
GasConsumption.count()                            : 359
DayStatistics.count()                             : 17
HourStatistics.count()                            : 337

--- Status (most recent) ---
DayStatistics                                     : 2016-12-26
ElectricityConsumption                            : 2016-12-27 21:34:00+00:00HourStatistics                                    : 2016-12-26 23:00:00+00:00GasConsumption                                    : 2016-12-27 20:00:00+00:00DsmrReading                                       : 2016-12-27 21:34:35+00:00`

dennissiemensma, 2016-12-27:
Man you really found some weird bugs :D I really hoped that one of the `(most recent)` stats would have hanged, but it all seems okay.

And it's only the **electricity graph** on the dashboard that 'hangs', or does the **gas** hang as well?

Jeltel, 2016-12-27:
Only electricity. 

And that's my job, right? 

Maybe I can set up a VPN to the pi. So you can snoop around yourself. It's already outside my main network. I'll try the openvpn solution I have seen earlier. Is there a private way to contact you?

dennissiemensma, 2016-12-27:
Well, that might be a solution, but I'm not sure whether I can find it as well. I hoped to be able to reproduce it immediately tonight.

Besides, the upcoming release is almost complete and it changes **a lot** regarding the parts you're having issues with:
- The dashboard graphs have been reworked (how it fetches data) and will be loaded a-sync (in background). 
- The statistics page fetches the data (that crashes) in a similar way. 
- The framework (Django) that is used for the database connection is upgraded as well. 

[The full changelog is located here](http://dsmr-reader.readthedocs.io/en/development/changelog.html). So it can be a fix, but since I cannot reproduce it, I'm not 100% sure.

So I can suggest to either wait for it's release, or you can try the pre-release yourself whether it fixes any of these issues. Then you would switch to the main `development` branch, which is almost release candidate (only #161 is still to be merged into it). 
Here at home I'm constantly running that branch on my 'production' environment, so it catches any bugs as well (so you won't be the only one using it).

The only thing is that it also causes your database to move 'forward' and **you can not switch back in one command**. 
Well you can, but you'd have to create a database backup just before upgrading and you can restore that in case it doesn't work at all. *But you'd lose any readings from the moment you upgrade to the restore (if you restore).*

So I think those are you options: wait a bit or upgrade early (please do not switch to the development branch right away).

Jeltel, 2016-12-28:
I would love to check it out. Maybe pick some hard to find bugs? ;)

But just switching to the development branch doesn't do everything I guess. What do I need to do?

Jeltel, 2016-12-28:
The recent electricity graph works again in version 1.5.

dennissiemensma, 2016-12-28:
Haha, well that's why I wrote *"please do not switch to the development branch right away"*, but I see you already found your way.

I hope you created or saved a database backup of before the upgrade, just to be sure.

dennissiemensma, 2016-12-28:
As it seems to work in the upcoming release, I won't put anymore effort into fixing it for `v1.4`. Thanks for trying and you patience.

closed 2016-12-28 by dennissiemensma