dsmrreader/dsmr-reader #180: Update contributors listing

dennissiemensma, 2016-12-15:
All contributors are currently listed over here: http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

But some are only mentioned by their nick name or (last) name:
* WatskeBart
* Sevickson

And these lack a link to their profile or linkedin:
* WatskeBart
* Sevickson
* Koen Volleberg

Give 'em a message if they want to have another name there or link to their linkedin.

dennissiemensma, 2016-12-15:
Hi @WatskeBart, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @Sevickson, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @kvolleberg, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @pvink, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @Helloitsme, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @goegol, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @Jeltel, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Hi @JeroenTielen, I'm currently updating the Credits-page for testers/contributors:
http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

* Do you (still) want to be mentioned there?
* If so, could you provide me your name or alias to list there?
* And optionally, do your want a link to your Github profile or LinkedIn-profile (if any) as well?

dennissiemensma, 2016-12-15:
Lots of spam, just state your preference below.

dennissiemensma, 2016-12-16:
In case there is any confusion:

> It's just a 'wall of fame' for the project. It doesn't make you a tester or demand that you test stuff. It's just a way to say 'thank you' for your contribution (which can be a simple as reporting an issue). It's similar to pages like these: https://sentry.io/thanks/
> 
> It's optional anyway, but just so you guy know. :]

goegol, 2016-12-17:
Hi! you can mentioned me.
You can use my full name: Hans-Paul Hemelaar and use my linkedin profile: https://www.linkedin.com/in/hanspaulhemelaar

JeroenTielen, 2016-12-18:
Hi Dennis,

That's nice :)

My name is: Jeroen Tielen
Linkedin: https://nl.linkedin.com/in/jeroentielen

Thanks. 👍🏻

dennissiemensma, 2016-12-18:
Thanks, it's noted.

kvolleberg, 2016-12-18:
Hi Dennis,
You can put my name on the list (Koen Volleberg).
I prefer not to have my LinkedIn account or my git account (which is
extremely empty) online.

Regards, Koen

dennissiemensma, 2016-12-18:
Sure, will do.

Jeltel, 2016-12-19:
You can add me:

https://nl.linkedin.com/in/jelteleijenaar

pvink, 2016-12-20:
Sure,

You can add me:
Paul Vinken
https://nl.linkedin.com/in/paul-vinken-934a7a44

dennissiemensma, 2016-12-21:
Done. Everyone has been added in the doc build for the next release. Development preview is here: http://dsmr-reader.readthedocs.io/en/development/credits.html#testers-contributors

dennissiemensma, 2016-12-22:
I've removed all my mentions above. 

Everyone mentioned before should at least have received an email regarding that mention, so it's up to everyone themselves to decided whether they are listed. :]

dennissiemensma, 2017-01-12:
New ticket will be created for future releases.

closed 2017-01-12 by dennissiemensma