dsmrreader/dsmr-reader #186: Super en duidelijk!

LadyIT, 2016-12-24:
Ga zo door met dit programma, top documentatie en het ziet er ontzettend goed uit. Ik ga het de komende tijd gebruiken en hoop op nieuwe features in de toekomst :-).

dennissiemensma, 2016-12-24:
Bedankt voor je feedback, het is fijn om te horen dat je er zo tevreden mee bent.

Wat betreft toekomstige features, in de **ontwikkelversie** van de documentatie:
* staat de [aankomende release](http://dsmr-reader.readthedocs.io/nl/development/changelog.html).
* staat al e.e.a. wat je [visueel kunt verwachten](http://dsmr-reader.readthedocs.io/nl/development/screenshots.html).

Wat betreft de features die je aandraagt:

_Combinatie grafiek van buitentemp en gasverbruik_
Bedankt voor je idee voor het samenvoegen van de temperatuur en gas. Ik ben wel zoekende naar een goede reden om dat te doen. Wil je toevallig de relatie weten tussen je gasverbruik en weersomstandigheden? Wellicht kun je dan beter [graaddagen](https://www.mindergas.nl/degree_days_calculation/explanation) gebruiken (houdt rekening met meer dan alleen temperatuur). Ik wilde dit eerst zelf inbouwen, maar MinderGas heeft er ook ondersteuning voor. 'Nadeel' is wel dat je daarvoor je gasstanden met hen moet delen, mits je daar geen bezwaar tegen hebt.

_in/uit zoomen in grafieken (zie voorbeeld domoticz p1 port uitlezing)_
Kun je toelichten in welke grafieken je dat graag ziet? Ik heb zelf geen ervaring met hoe dit werkt in domoticz. Welliswaar dat het mogelijk is, maar dat ligt erg aan hoe het werkt, wat je verwacht en vooral waarvoor het gebruikt wordt. 
Alle data in de applicatie wordt namelijk uiteindelijk afgevlakt naar uur- en dagaggregaties, dus dat kan ook nog een mogelijke beperking zijn.

LadyIT, 2016-12-25:
Hi Dennis,

Bedankt voor de tip over de graaddagen, dit is inderdaad wat ik bedoel. Mindergas kan hier inderdaad een uitkomst in bieden. Uploaden gegevens geen probleem, maar een "alles in 1 interface" is natuurlijk net wat fijner..

In/uitzoomen grafiek:
DSMR Reader geeft nu real time een half uur weer met een interval van 1 minuut. Misschien een idee om dit op te rekken naar de "laatste 24 uur" net zoals het gasverbruik, en dat de gebruiker dan zelf tot een maximale verschaling van 5 minuten (of 1 minuut zoals nu) kan inzoomen. 
[image]

dennissiemensma, 2016-12-25:
Bedankt voor je input en het voorbeeld!

Wat betreft de graaddagen. Ik begrijp helemaal dat het handig is om alles in 1 applicatie of grafiek te hebben. Alleen is dat niet mijn expertise én vereist het onderhouden van de graaddagen ontzettend veel werk. 
Het KNMI upload namelijk alle graaddagen met terugwerkende kracht op (meestal) de eerste dag van de maand. Die gegevens moeten dan ergens vandaan gehaald wordt en verrekend worden met de gemeten gegevens thuis. Ik laat dat dan liever over aan een partij die zich dat eigen heeft gemaakt, zoals MinderGas.

Voorlopig zal ik dit om die redenen niet inbouwen. Dit project probeert zich zoveel mogelijk te beperken op het uitlezen van de slimme meter en plotten van die data. Een ander verzoek met het uitlezen van een SMA-omvormer heb ik overigens om dezelfde reden gesloten: #145

Wat betreft de grafieken, ik zal kijken of ik zoiets inzichtelijk kan maken. In de volgende release komt sowieso al een verbetering via #164, waarbij wat verder terug te scrollen is dan het laatste halfuur.

dennissiemensma, 2016-12-25:
Voor het laatste heb ik nu #189 aangemaakt. Mocht je nog meer suggesties hebben, vraag maar raak. Alle input is welkom. Nogmaals bedankt :]

closed 2016-12-25 by dennissiemensma

LadyIT, 2016-12-26:
Hi Dennis,

Helemaal duidelijk, thnx voor je snelle antwoorden en acties :). #164 is een mooie toevoeging inderdaad!

dennissiemensma, 2017-01-01:
@LadyIT  de [`v1.5` release is uit](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0).

Lees hiervoor het volgende:
- [release notes](http://dsmr-reader.readthedocs.io/nl/latest/changelog.html#v1-5-0-2017-01-01) 
- [how-to-upgrade docs](http://dsmr-reader.readthedocs.io/nl/latest/faq.html#how-can-i-update-my-application)

N.B.: Voor het upgraden moet je, nadat je inglogd bent als `dsmr` gebruiker, cd doen naar de juiste map én de VirtualEnv activeren.
Dit stond eerder wat minder prominent in de documentatie, [maar dat is nu aangepast](http://dsmr-reader.readthedocs.io/nl/latest/installation.html#virtualenv).