dsmrreader/dsmr-reader #188: Verify telegrams' CRC

dennissiemensma, 2016-12-24:
The telegrams should be verified by checking their CRC. I'm planning to use #154 but that also lacks CRC checks, for now.

I should either re-check later or write it myself.

dennissiemensma, 2016-12-25:
Requested via #187

ThinkPadNL, 2016-12-25:
Too bad it doesn't use CRC yet. I have had some incorrect data a few times, so it is definately a recommendation to use, if you have a modern enough meter (>= DSMR4). I now use some other library and he has implemented CRC checking: https://github.com/nrocco/smeterd/issues/10
Maybe you can find some inspiration there? The method used is `crc16`.

dennissiemensma, 2016-12-25:
@ThinkPadNL a lot of thanks to you for providing the info! I managed to have the CRC checksum working now. 

I will run it at home for a while and merge fbaed6ca1f7046461a42db35564a829e6473b5a9 back to the main `development` branch for the upcoming release when no problems occur.
That release is [scheduled for mid January](https://github.com/dennissiemensma/dsmr-reader/milestone/20), so this feature won't be available right away.

Optionally, if you'd like to be listed on the 'thanks' page, please leave a note in #180.

ThinkPadNL, 2016-12-25:
Haha, you added that very quickly! Great!

Nah, i don't have to be listed as i didn't make it. 'nrocco' from that GitHub page is to be mentioned if you used his code for inspiration :)

dennissiemensma, 2016-12-25:
Sure, I will mention him and drop him a message. 

I won't list you if you do not want to. Please note that I do not list people for providing other's code, but for bringing the ideas and pointing out any issues. Just so you know. :]

dennissiemensma, 2016-12-25:
Code is running fine for few hours now, although no bad telegrams were seen.

The [tests pass](https://travis-ci.org/dennissiemensma/dsmr-reader/builds/186689007) as well. So merged back to `development` for `v1.5`.

Thanks again for requesting this feature.

closed 2016-12-25 by dennissiemensma

dennissiemensma, 2017-01-01:
@ThinkPadNL the [`v1.5` release is out](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0), containing this issue.

Please mind:
- [the release notes](http://dsmr-reader.readthedocs.io/en/latest/changelog.html#v1-5-0-2017-01-01) 
- [how-to-upgrade docs](http://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-update-my-application)

Also note that upgrading requires you to cd into the project root and activate the VirtualEnv, after switching to the `dsmr` user. The installation guide hinted that before, but it's [highlighted more clearly now](http://dsmr-reader.readthedocs.io/en/latest/installation.html#virtualenv).

ThinkPadNL, 2017-01-01:
I installed your software today, now that it has CRC-checking (i have a 5m long cable from my meter cabinet to my server, so i needed CRC to ignore false data. #187 would let me remove the long cable ).

Where can i see how many corrupt packages it ignored?

dennissiemensma, 2017-01-01:
You can see it on the Statistics page, but it only displays the counter when any readings were discarded. The counter can also be seen in the Admin (Configuration) at the `Meter statistics` menu item. 

It contains several fields, and one of them is `rejected_telegrams`: `Number of rejected telegrams due to invalid CRC checksum` (or translated). It should always be listed there.

ThinkPadNL, 2017-01-01:
`Rejected telegrams: 0`. Software is running for about 40 minutes now.
I guess that it was a wise decision to use some twisted CAT5E-cable to create a P1-USB cable.

dennissiemensma, 2017-01-01:
Cool, I'd love to see any reading fail though. I could only reproduce a failed CRC manually while writing mocked tests so far. Not in production use.

Regarding the language, you can pick either English or Dutch. I'm actually surprised, and also a bit honored, that people just seem to start talking English here anyway. Good ol' Dutch habbits I guess.

ThinkPadNL, 2017-01-01:
There it is:
`Rejected telegrams: 1`

Graphs are still looking good (i don't see any enormous usage as i had with my own script), so i can conclude (with this real life example) that the verification of the telegram checksum is working correctly!

dennissiemensma, 2017-01-01:
Mocht je trouwens nog benieuwd zijn naar het foutieve telegram, volgens mij kan je hem hiermee vinden:

`grep InvalidTelegramChecksum /var/log/supervisor/dsmr_datalogger.log -C 20`

En anders zou het [een van de twee foutmeldingen in deze code](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.0/dsmr_datalogger/services.py#L94) moeten zijn (vermoedelijk de laatste).