dsmrreader/dsmr-reader #189: Dashboard graphs zooming

dennissiemensma, 2016-12-25:
I should investigate whether it's possible to implement zooming in the Dashboard graphs. The x-range could be something between 30 minutes and 24 hours (similar to gas).

[Reported by @LadyIT in #186]

Jeltel, 2016-12-28:
A viewing window of just an hour is a bit short. I would like to see a way to switch the viewing window of the recent electricity graph. Like one hour per minute (like it is implemented now), three hours per minute (a bit much on a phone screen, but great on pc) and 24 hours per hour.

dennissiemensma, 2017-01-17:
Om even alles te inventariseren (voor mijn overzicht) zijn er per onderdeel nu nieuwe verzameltickets. Voor dit ticket is dat vooral #235.

Daarin kijk ik in de eerste instantie wat er mogelijk is voor een eventuele proof of concept. Eventuele input of feedback kan dan later op gegeven worden. Niet alle wensen van iedereen zijn overigens doorgaans even haalbaar of praktisch.

ThinkPadNL, 2017-03-27:
Enig idee wanneer je deze/235 kunt oppakken? Onlangs verhuisd en tijdje zonder slimme meter gezeten, inmiddels weer eentje hangen. Ontbreken van dit grafiektype is toch wel iets waar ik vrij gauw na installatie `dsmr-reader` tegenaan liepen. Ook #236 is eentje die ik graag zou zien :D

dennissiemensma, 2017-03-27:
Zie mijn reactie in #213, idem voor hier.

dennissiemensma, 2018-03-05:
Ik ben overigens laatst wel begonnen aan een poc voor een redesign, maar het viel me behoorlijk tegen om dat werkend te krijgen, zonder te veel functionaliteit te verliezen. 
Het kan zijn dat ik voor een toekomstige versie de GUI op de schop gooi, maar voor het gros van de doelgroep is de huidige opzet afdoende, al kan het zeker 10x mooier/netter/efficienter.

closed 2018-03-05 by dennissiemensma

dennissiemensma, 2018-07-11:
Inmiddels heeft de laatste release op het dashboard zoombare grafieken via #489.