dsmrreader/dsmr-reader #193: EOF error on statistics page

Jeltel, 2016-12-27:
When opening the statistics page the nginx server returns an error:

` Server Error

Exception:

OperationalError: SSL SYSCALL error: EOF detected
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/handlers/base.py", line 132, in get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 89, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/statistics.py", line 33, in get_context_data
    context_data['slumber_consumption_watt'] = dsmr_consumption.services.calculate_slumber_consumption_watt()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/services.py", line 218, in calculate_slumber_consumption_watt
    if not most_common:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 166, in __bool__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 965, in _fetch_all
    self._result_cache = list(self.iterator())

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 1085, in iterator
    for row in self.query.get_compiler(self.db).results_iter():

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 794, in results_iter
    results = self.execute_sql(MULTI)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 840, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 98, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 64, in execute
    return self.cursor.execute(sql, params)
`

The weinterface fails after this error:
`Server Error

No exception or traceback available.`
After a few minutes everything but the statistics page works fine again.
Reboot doesn't solve the issue

dennissiemensma, 2016-12-27:
Thanks for reporting this and providing the traceback as well. I will look into it after checking out your other issue.

dennissiemensma, 2016-12-27:
I have never seen this error before. Both Google and the traceback point at some problem with PostgreSQL, which sends the error `SSL SYSCALL error: EOF detected` back to the application.

**I'm seeing a line containing the word `VM`, are you running the application in a VM or just native on a RaspberryPi?**

Someone [mentioned a lack of memory available for PostgreSQL](http://stackoverflow.com/a/36086325) once, having the same error. The `slumber_consumption_watt` method, the code seems to crash on, is one that requests PostgreSQL to calculate the average on **all** readings taken, so that might require some memory.

However, there seem to be many other causing this error as well, so please elaborate on what setup you are running exactly. :]

Jeltel, 2016-12-27:
I'm using a raspberry pi 2 with raspbian Jessie via NOOBS. With nothing but pi-hole and dsmr running.

Updated and upgraded last weekend.

dennissiemensma, 2016-12-27:
Okay, can you check the available memory, by typing on the command line `sudo free -m`?

Jeltel, 2016-12-27:
`pi@pi-hole:~ $ sudo free -m
             total       used       free     shared    buffers     cached
Mem:           925        303        622         35         21        141
-/+ buffers/cache:        139        785
Swap:           99          0         99`

dennissiemensma, 2016-12-27:
That seems fine. Lot's of unused memory, so it's another issue.

I continued to look for reasons (it's a bummer I cannot reproduce this as well, so I can check any fix first). Some issues were around the database driver this project uses.

But a fix was released and that version is already used by this application (since `v1.3+`).
```
Psycopg 2.6.2 released
- Fixed inconsistent state in externally closed connections
  (:tickets:`#263, #311, #443`).
```

But can you check for sure with:
* `su - dsmr`
* `pip freeze | grep psy`

It should list the version used.

dennissiemensma, 2016-12-27:
I can also be caused by the database connection caching used by default.

You can try to disable it by:
- `su - dsmr`
- Edit `dsmrreader/settings.py`
- Remove this line: `'CONN_MAX_AGE': 300,`
- Save the file.
- Go back to root/sudo user and restart the `dsmr_webinterface` process in Supervisor.

Jeltel, 2016-12-27:
On the pip/grep I don't get a result. 

And removing the max age line didn't help.

dennissiemensma, 2016-12-27:
The pip/grep should work, but I think it has the same cause as in the other ticket and [can be fixed this way](https://github.com/dennissiemensma/dsmr-reader/issues/178#issuecomment-269385470).

Jeltel, 2016-12-27:
It does work now with the bash file: 

`(dsmrreader)dsmr@pi-hole:~/dsmr-reader $ pip freeze | grep psy
psycopg2==2.6.2`

So, correct version

Jeltel, 2016-12-28:
This is also not present in version 1.5

dennissiemensma, 2016-12-28:
Great, still weird, but great that it seems to be resolved. Thanks for testing.

closed 2016-12-28 by dennissiemensma