dsmrreader/dsmr-reader #194: [Request] Add timestamp to highest and lowest Watt occurance

dbmnl, 2016-12-28:
I have been using this project for a couple of weeks now and it's working perfectly. So kudos to Dennis for making this so we all can enjoy it.

I do however want to make a suggesting to the statistics page. Currently it shows you the highest and lowest amount of Watts used in the history of the code being alive on your Pi. However it does now say "When" this moment was. If you want to track back to why you are using to much energy, Or if you want to know what you did to get the lowest amount of energy you need a timestamp so you can search further.

It would be nice if the timestamp is added to the page. So that you can continue your investigation of the source.

I created a little mockup from what i mean.

[http://oi64.tinypic.com/161c3s4.jpg](url)

Jeltel, 2016-12-28:
That would be handy yes. And even more great would be a link to a graph showing the measurements around the min/max. Just to see if it was a one time thing.

Also, here's the mockup :)

[image]

dennissiemensma, 2016-12-28:
Hi @dbmnl thank you for your kind words.

I think your request makes sense, so I've added it to the backlog of the `v1.6` release, which is scheduled four to six weeks from now. `v1.5` will be release first, soon.

However, I can already provide you that info, as it's already stored in your database. In case you have chosen `PostgreSQL` as database backend, please execute this command/queryas root/sudo-user on the commandline:

```
sudo -u postgres psql dsmrreader -c "select read_at, currently_delivered from dsmr_consumption_electricityconsumption order by currently_delivered desc limit 25;"
```

It will list your the timestamps and usage of the 25 highest usage peaks read. For example:
```
        read_at         | currently_delivered 
------------------------+---------------------
 2016-11-19 18:39:00+01 |               6.895
 2016-10-27 17:00:00+02 |               6.388
 2015-12-20 17:07:00+01 |               6.381
 2016-08-19 21:15:00+02 |               6.231
 2015-12-20 17:05:00+01 |               6.216
 2016-08-19 21:14:00+02 |               6.189
...
```

dbmnl, 2016-12-29:
Great tnx Dennis. It seems christmas caused a lot of high usage :). Note to self, do not invite family anymore! Also having two ovens and using induction does not help offcourse. Diner was delicious though :P

        read_at         | currently_delivered
------------------------+---------------------
 2016-12-26 17:09:09+00 |               8.831
 2016-12-26 17:11:09+00 |               8.731
 2016-12-26 17:15:10+00 |               8.687
 2016-12-26 17:13:09+00 |               8.645
 2016-12-26 17:13:19+00 |               8.042
 2016-12-26 17:09:19+00 |               7.939
 2016-12-26 17:14:59+00 |               7.868
 2016-12-25 16:25:40+00 |               7.698
 2016-12-25 16:25:30+00 |               7.627
 2016-12-26 17:12:59+00 |               7.497
 2016-12-25 07:12:13+00 |               7.348
 2016-12-25 07:12:43+00 |               7.270
 2016-12-25 07:12:33+00 |               7.244
 2016-12-25 07:12:53+00 |               7.217
 2016-12-25 07:13:03+00 |               7.196
 2016-12-25 07:13:13+00 |               7.194
 2016-12-25 07:12:23+00 |               7.090
 2016-12-26 17:23:10+00 |               7.066
 2016-12-26 17:10:59+00 |               7.055
 2016-12-25 16:27:20+00 |               6.826
 2016-12-25 16:27:50+00 |               6.775
 2016-12-25 16:28:00+00 |               6.766
 2016-12-25 16:27:10+00 |               6.661
 2016-12-26 17:08:59+00 |               6.638
 2016-12-25 16:27:40+00 |               6.609

dennissiemensma, 2016-12-30:
Sure, you're welcome. 

I think I'll refactor this feature a bit in the future. Because now you will only see new highs when a future reading exceeds the top one listed now. 

It'd be better to have this as a separate category in the Export page. And then I can easily extend it to have another option for the phases ([such as requested and deferred manually here](https://github.com/dennissiemensma/dsmr-reader/issues/161#issuecomment-268869066)). The biggest plus is that you can also select a complete date range there. In case you want to only see the highs of, let's say, 2017.

To be continued.

dbmnl, 2017-03-19:
Hi Dennis,

I was thinking some more about this and i think this requirement is still valid however i now have some additions to them :)

My initial request is to only display the date/time when the highest lowest occurance was there. However I now think this is not enough.

With the addition of Phases in 1.5 we now have a world of information available to maybe extend the daily pages (under archive) a bit to show the following items per day:

1. Lowest Watt occurance [Total] - Date/Time
2. Highest Watt occurance [Total] - Date/Time
3. Most used Watt occurance [Total]
4. Lowest Watt occurance [Phase 1] - Date/Time
5. Highest Watt occurance [Phase 1] - Date/Time
6. Lowest Watt occurance [Phase 2] - Date/Time
7. Highest Watt occurance [Phase 2] - Date/Time
8. Lowest Watt occurance [Phase 3] - Date/Time
9. Highest Watt occurance [Phase 3] - Date/Time

The benefit of this is that when you want to cut down Idle Watt usage you can see on a daily basis if this is going the right way, or if you have some appliances which are using too much. By displaying the high/low values of the phases you can narrow down which appliances are causing the issue and thus lowering your powerconsumption easier.

dennissiemensma, 2017-03-19:
Thanks for sharing the ideas. I will consider this for a future version. It does makes more sense to display these kinds of statistics in the Archive page, since they have a limited time scope anyway.

dennissiemensma, 2017-10-08:
Ik heb een opzet gemaakt om de gevraagde gegevens te tonen. Daarbij heb ik de 'laagste' waarden voor L1, L2 en L3 wel weggelaten. Deze bleken namelijk vaak `0` terug te geven (of een gelijke aan het totaal). 

Hieronder een voorbeeld:
[image]

Later kan dit eventueel nog wat bijgewerkt worden als dit niet genoeg helderheid verschaft.

dennissiemensma, 2017-10-08:
Gaat mee met v1.10

closed 2017-10-08 by dennissiemensma

dbmnl, 2017-10-09:
Hey Dennis,
Ziet er super uit en inderdaad precies wat ik verwacht. Tnx voor het mogelijk maken hiervan. Ik wacht met smart op 1.10 :)