dsmrreader/dsmr-reader #195: V 1.5: Recent Graph Scrolling

Jeltel, 2016-12-28:
Just updated to development branch version 1.5.

Nice option to browse through the recent measurements up to a week ago! I do have a few remarks:

- You can see back for a week, but there isn't a way to see what day I'm watching. Can you implement a date somewhere?
- The scroll buttons shoot below each other when using a phone, taking up a lot of screen space (weird spacing is also seen in other places, check dsmr on a phone);
- A viewing window of just an hour is a bit short. I would like to see a way to switch the viewing window of the recent electricity graph. Like one hour per minute (like it is implemented now), three hours per minute (a bit much on a phone screen, but great on pc) and 24 hours per hour.

That would be great :)

dennissiemensma, 2016-12-28:
Hi, thank you for trying the upcoming release. Do all the other pages work fine as well? 

I've just promoted the `development` branch to the 'release candidate' flow, since the last issue was merged back. You can update easily by executing `./deploy.sh` in the root of the project folder (as `dsmr` user).

dennissiemensma, 2016-12-28:
Regarding your remarks:
* I will try to see if I can fix the first two. Or improve it a bit.
* Your last remark seems to have this recent feature request in common: #189

dennissiemensma, 2016-12-28:
It's hard to develop an interface for both desktop and mobile, but I think I've managed to improve the scroll buttons a bit. Thankfully the Bootstrap framework really helps with that.

You can update (again), it should be better since `v1.5rc2` (see top right corner).

Jeltel, 2016-12-28:
As far as I can see, the other pages work fine. Also the buttons are better. 
Thanks :)

And yes #189 seems somewhat similar. I'll post it there as well.

dennissiemensma, 2016-12-30:
As stated before, I've fixed the second request and the third one should be relayed to the other ticket.

The first request, regarding indicating the date, will be postponed to either `v1.6` or be released in a `v1.5.x` patch. 
For now, I will accept that it's not perfect yet, but won't have it delay the `v1.5` release. Because it's not very trivial to fix.

Jeltel, 2016-12-30:
That's ok. I'll try to find a screenshot of another implementation of a graph which uses the date selector pretty well.

dennissiemensma, 2016-12-30:
I appreciate it but it's not required. :]

I've already [found a plugin](https://github.com/chartjs/chartjs-plugin-annotation) for the chart library used (chartjs). But implementing this also requires the backend to mark an X-axis value as a 'day transition', in order to render the requested line when needed. 
And that's the hardest part, to integrate it, because since `v1.5` your browser simply requests the server the data it's looking for. 

You can view it yourself by going to your Dashboard and append `/xhr/graphs?units_offset=0` to the URL in your browser. It has to be communicated through there, which takes some time.

dennissiemensma, 2016-12-30:
@Jeltel I think I have just solved this more easily after all.

In `v1.5rc4` the x-axis will now simply **also** list the day, but only when scrolling. You can try it yourself by updating using the `./deploy.sh` script.

Jeltel, 2016-12-30:
Weekday is good enough for me. Well done. 

You also have resized the graphs a bit, for the better!

dennissiemensma, 2017-01-01:
Thanks. Then I won't put any more effort in improving this. Seem a good enough fix for me as well.

closed 2017-01-01 by dennissiemensma

dennissiemensma, 2017-01-01:
@Jeltel the [`v1.5` release is out](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0).

You should switch back to the `master` branch:

* `sudo su - dsmr`
* `git pull`
* `git checkout master`
* `git pull` again, to move your local master (v1.4.1) forward to the latest master version (v1.5).

And finally:
* Execute `./deploy.sh` to make sure everything still runs fine. The application will shortly run old `v1.4.1` code between the `git checkout master` and second `git pull` command. This will reload it to make sure it uses the latest.

You should see `v1.5` (without rc) in the top right corner of the application.

Jeltel, 2017-01-01:
There is some issue:

`dsmrreader)dsmr@pi-hole:~/dsmr-reader $ git checkout master
error: Your local changes to the following files would be overwritten by checkout:
        .travis.yml
        README.md
        deploy.sh
        docs/_static/screenshots/archive.png
        docs/_static/screenshots/compare.png
        docs/_static/screenshots/dashboard.png
        docs/_static/screenshots/export.png
        docs/_static/screenshots/mindergas_admin.png
        docs/_static/screenshots/statistics.png
        docs/_static/screenshots/status.png
        docs/_static/screenshots/trends.png
        docs/api.rst
        docs/application.rst
        docs/changelog.rst
        docs/conf.py
        docs/contributing.rst
        docs/credits.rst
        docs/faq.rst
        docs/installation.rst
        docs/intro.rst
        docs/licence.rst
        docs/locale/nl/LC_MESSAGES/api.mo
        docs/locale/nl/LC_MESSAGES/api.po
        docs/locale/nl/LC_MESSAGES/application.mo
        docs/locale/nl/LC_MESSAGES/application.po
        docs/locale/nl/LC_MESSAGES/changelog.mo
        docs/locale/nl/LC_MESSAGES/changelog.po
        docs/locale/nl/LC_MESSAGES/contributing.mo
        docs/locale/nl/LC_MESSAGES/contributing.po
        docs/locale/nl/LC_MESSAGES/credits.mo
        docs/locale/nl/LC_MESSAGES/credits.po
        docs/locale/nl/LC_MESSAGES/faq.mo
        docs/locale/nl/LC_MESSAGES/faq.po
        docs/locale/nl/LC_MESSAGES/index.mo
        docs/locale/nl/LC_MESSAGES/index.po
        docs/locale/nl/LC_MESSAGES/installation.mo
        docs/locale/nl/LC_MESSAGES/installation.po
        docs/locale/nl/LC_MESSAGES/intro.mo
        docs/locale/nl/LC_MESSAGES/intro.po
        docs/locale/nl/LC_MESSAGES/licence.mo
        docs/locale/nl/LC_MESSAGES/licence.po
        docs/locale/nl/LC_MESSAGES/screenshots.mo
        docs/locale/nl/LC_MESSAGES/screenshots.po
        docs/screenshots.rst
        dsmr_api/tests/test_api.py
        dsmr_backend/management/commands/dsmr_backend.py
        dsmr_backend/mixins.py
        dsmr_backend/services.py
        dsmr_backend/tests/test_backend.py
        dsmr_backend/tests/test_services.py
        dsmr_backup/services/backup.py
        dsmr_backup/services/dropbox.py
        dsmr_consumption/fixtures/dsmr_consumption/test_dsmrreading.json
        dsmr_consumption/fixtures/dsmr_consumption/test_dsmrreading_without_gas.json
        dsmr_consumption/migrations/0001_squashed_0004_recalculate_gas_consumption.py
        dsmr_consumption/models/consumption.py
        dsmr_consumption/services.py
        dsmr_consumption/tests/test_services.py
        dsmr_datalogger/dsmr.py
        dsmr_datalogger/models/reading.py
        dsmr_datalogger/models/settings.py
        dsmr_datalogger/services.py
Please, commit your changes or stash them before you can switch branches.
error: The following untracked working tree files would be overwritten by checkout:
        .hgignore
        docs/_static/screenshots/configuration.png
        dsmr_consumption/migrations/0001_initial.py
        dsmr_consumption/migrations/0002_split_dsmr_reading_fields.py
        dsmr_consumption/migrations/0003_realign_gas_read_at.py
        dsmr_consumption/migrations/0004_recalculate_gas_consumption.py
Please move or remove them before you can switch branches.
Aborting
(dsmrreader)dsmr@pi-hole:~/dsmr-reader $`

What to do?

dennissiemensma, 2017-01-01:
Can you try `git checkout -f master`?

Jeltel, 2017-01-01:
`(dsmrreader)dsmr@pi-hole:~/dsmr-reader $ git checkout -f master
error: unable to create file dsmr_datalogger/migrations/0001_initial.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/0001_squashed_0005_optional_gas_readings.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/0002_add_meter_statistics.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/0002_meta_names.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/0003_split_dsmr_reading_fields.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/0004_multiple_dsmr_version_support.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/0005_optional_gas_readings.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/migrations/__init__.py (Structure moet worden opgeschoond)
error: unable to create file dsmr_datalogger/tests/__init__.py (Structure moet worden opgeschoond)
fatal: cannot create directory at 'dsmr_datalogger/tests/datalogger': Structure moet worden opgeschoond
(dsmrreader)dsmr@pi-hole:~/dsmr-reader $`

dbmnl, 2017-01-01:
It looks like your SD card is corrupt.... Had the same messge before regarding the cleaning of the structure.... you can try a few google searches on the errormessage to solve it. but in my case it was a lost cause.

Jeltel, 2017-01-01:
Hmm. The issue with raspberry pi's.

dennissiemensma, 2017-01-01:
Hmmm, I asume you've tried rebooting and all? Would be a shame if it's corrupt again, since you've had that shortly ago as well (#178). 

If the card really is corrupt again, you can try [the same I'm using](https://www.dataio.nl/samsung-16gb-micro-sd-evo-met-adapter-uhs-i-class-10-48mbs/), which isn't very expensive.

Another option is to run the application off the raspberrypi, but that requires you to either have a VPS or server somewhere as well.

Jeltel, 2017-01-01:
I did. But somehow restarting again solved it?

But it is good advice to get a proper SD card.

Jeltel, 2017-01-04:
Bought that SD-card, too bad its just a tiny bit smaller. I can't just transfer a complete image to the new SD-card. A well, a fresh start!

dennissiemensma, 2017-01-04:
@Jeltel ooh, you can just order any other size you want. It's just the brand/type I wanted to mention.

Samsung EVO seems to perform well. It runs for over a year here now, while having 'consumed' multiple cheap cards already (me, family, colleagues). The same applies to some of Sandisk's cards, but I haven't used them myself.

But it's still a hardware issue and most SD cards aren't made to be written to by a database. So I guess it's always about having a bit of luck, hence the builtin backups to keep data loss to a minimum :]

Jeltel, 2017-01-04:
Well why not putting the db on a HDD?
I have a spare one which in intent to use for nextloud.

And I ordered a 16gb, which is the same as I had. But apparently it's a few thousand sectors larger ;). A well, I was up and running pretty fast. And I made an image just after completing.

dennissiemensma, 2017-01-04:
Yeah, if needed, you can put the database anywhere. 

Depending on where it's connected to I can even help you with that. If you mount it directly on your Pi you could simply:

- stop the database
- move/rename `/var/lib/postgresql`
- mount `/var/lib/postgresql` on HDD
- move the contents of the renamed folder above back to  `/var/lib/postgresql`
- start the database again

Jeltel, 2017-01-04:
I might try that.
With the Samsung SD card it's way faster. Recommended 'investment'. 👍