dsmrreader/dsmr-reader #196: [Request] Overview of meter readings per day on a specified time

dbmnl, 2016-12-29:
Before i was using this project I kept track of energy / gas usage through an excel sheet which is having many years of data in it. In order for this to work i walk to the meter to write down the numbers every day. Now that we have this project this can offcourse be automated :-)

What would help in this is we would have a page in the project (or underneath the page archive) to easily see the meter readings of a particular day entire week or entire month on a day to day basis. THis way these can be taken over to a seperate excelsheet or be given to an energycompany if requested. 

I myself always write them down around 08:00.  With the query from my last request i dived into the DB to get an overview myself already using the commandline but it would offcourse be great to get this info from the webpage. (in order to track a certain timepoint i used the meters logic that it will only sent a telegram once every 10 seconds which will result in only one telegram per day)

**Nice to have extra's**

- Configure a specific time from when these readings need to be taken (e.g. 08:00 in stead of 0:00) (configuration can be done from the configuration screen)
- Send these values to your mobile phone using the notification service ( this will require the same as the extra request above to sent it on a particular time e.g. 08:00 in stead of 0:00 now) (so you do not even need to open the webpage!
- Create a seperate page for the readings as shown below.
- Add the information to the achive page when clicking on  a specific day what the meter readings were for that day on a specified time for easy tracking back.

`sudo -u postgres psql dsmrreader -c "select distinct read_at::date, delivered_1, delivered_2 from dsmr_consumption_electricityconsumption where read_at::time between time '08:00:00' and '08:00:10' order by read_at::date desc;"`

```
read_at   | delivered_1 | delivered_2
------------+-------------+-------------
 2016-12-29 |     438.550 |     330.858
 2016-12-28 |     437.676 |     324.973
 2016-12-27 |     436.937 |     317.567
 2016-12-26 |     425.071 |     317.167
 2016-12-25 |     410.191 |     317.167
 2016-12-24 |     393.098 |     317.167
 2016-12-23 |     392.202 |     313.392
 2016-12-22 |     390.411 |     310.608
 2016-12-21 |     389.757 |     307.400
 2016-12-20 |     389.214 |     303.642
 2016-12-19 |     388.694 |     300.601
 2016-12-18 |     378.652 |     300.014
 2016-12-17 |     373.599 |     300.014
 2016-12-16 |     372.953 |     297.630
 2016-12-15 |     372.433 |     295.385
 2016-12-14 |     370.674 |     292.790
 2016-12-13 |     370.173 |     289.998
 2016-12-12 |     369.577 |     286.719
(18 rows)
```

dennissiemensma, 2016-12-30:
Hi, thank you for your (very well explained) idea. I'm also happy to see that you already managed to extract quite some information from the database yourself, which is nice because it' one of the goals of this project (keep your data for yourself and play with it).

Regarding your request for these extra's: I really like them and I totally see the purpose you have for them, but I'm not sure whether I should (or will) implement them all. Or at least in some form.

**Sending the meter positions daily to your phone for a specific time**
Not sure about this, but I might merge it with the existing daily notification, but extending that message with any meter positions. And also allowing for a different time of sending the notification (which is after 1 a.m. now).

**Separate page for readings**
I personally think that what you are asking here is more of a feature for the Export page. It already allows (date) selections, but currently only applies to the day & hour statistics. Having a new export option of the meter positions (on a specific time) seems better to me than introducing a new page for it.

**Add the information to the Archive page**
I don't think this is possible, simply due to the fact that the Archive page does not support a time selection. Also, you should be able to accomplice the same using a new Export option discussed in the paragraph above.

dbmnl, 2017-01-01:
Great to hear dennis. Tnx for looking at implementing this. 
As for the notification service, I really like the idea to see everything on your mobile at a glance. Especially since most people these days only use mobile phones having features towards this platform will be most welcomed by most people (or well by me atleast :P) 

Since i doubt your scope will be to create an app for this you might already have a thought about extending the notification services to for instance,

- Sending a notification if your power usage exceeds your total configured "aansluitwaarde" (3x25A, 1x25A etc) (especially can be an issue for homes still connected to 1x25A)
- When Phase consumption is added in V1.5 sent notification if individual phase is exceeding it's configured"aansluitwaarde" (3x25A, 3x35A etc). 
- Sent notification if Gasusage is higher than average usage 
- etc etc

Many things can be thought of which can alert the user making the user "more" aware of it's power consumption.  

Please note, i am just thinking out loud here :) So no need to actually do things with my comment. I am just a happy user :)
- Sending notification if a set limit

Martinvdm, 2017-12-30:
Perhaps a good idea to make a custom MQTT mapping for the daily used.

dennissiemensma, 2018-03-05:
Voor de specifieke wensen in dit ticket, qua meterstanden rond 8:00. is het denk ik handiger om dit simpelweg via een cronjob te doen. Dus een psql query en dan de output pipen naar het mail command in Linux. Heel basic, maar dan heb je wel een historie van de meterstanden op dat tijdstip.

Ik ga nog wel de meterstanden opnemen in de dagstatistieken, maar daar ga ik geen onderscheid in maken in het tijdstip. Vermoedelijk worden dat gewoon de meterstanden van middernacht.

closed 2018-03-05 by dennissiemensma