dsmrreader/dsmr-reader #197: No gas readings

markruys, 2016-12-29:
Awesome application, installation of a Debian Jessie BeagleBone went smooth. Thanks a lot!!

What I don't understand is how to enable gas readings. They stay at 0 m3, although the P1 does pass gas usage:
```
/XMX5LGBBFFB231322517

1-3:0.2.8(42)
0-0:1.0.0(161229185431W)
0-0:96.1.1(************************)
1-0:1.8.1(003521.808*kWh)
1-0:1.8.2(003256.607*kWh)
1-0:2.8.1(000383.285*kWh)
1-0:2.8.2(000920.962*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(06.593*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00002)
0-0:96.7.9(00001)
1-0:99.97.0(1)(0-0:96.7.19)(151015165429S)(0002320720*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(028*A)
1-0:21.7.0(06.593*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(**************)
0-1:24.2.1(161229180000W)(01917.816*m3)
!CA07
```

markruys, 2016-12-29:
I had to be a little more patient...

dennissiemensma, 2016-12-30:
It's okay. I don't think that this is explained anywhere in the docs anyway, so I might just add a little note regarding this, for others. :]

dennissiemensma, 2016-12-30:
I've updated the FAQ for the docs of the upcoming release in e90fc4108b315fceae1690641d7516c19b3d49f6. Thank you for indicating this issue.

dennissiemensma, 2017-01-01:
@markruys  the [`v1.5` release is out](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.5.0).

Please mind:
- [the release notes](http://dsmr-reader.readthedocs.io/en/latest/changelog.html#v1-5-0-2017-01-01) 
- [how-to-upgrade docs](http://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-update-my-application)

Also note that upgrading requires you to cd into the project root and activate the VirtualEnv, after switching to the `dsmr` user. The installation guide hinted that before, but it's [highlighted more clearly now](http://dsmr-reader.readthedocs.io/en/latest/installation.html#virtualenv).