dsmrreader/dsmr-reader #198: Bij wegvallen internetverbinding verspringt buitentemp

LadyIT, 2016-12-31:
Puntjes op de i : 
Als het internet wegvalt en na een x aantal uur weer opkomt, dan zit er een sprong in de X-as (tijd) in de buitentemperatuur grafiek. Mogelijk om de verloren gegevens in dat tijdbestek terug op te halen en/of de x-as gewoon doorlaten lopen zonder gegevens oid? In het voorbeeld hieronder tussen 15:00 en 19:00.

[image]

dennissiemensma, 2017-01-01:
Bedankt voor je ticket.

Het is niet mogelijk om de gegevens met terugwerkende kracht op te halen. De applicatie gebruikt [deze pagina van Buienradar](http://api.buienradar.nl/) en die toont alleen actuele gegevens.

Ik kan wel kijken of het mogelijk is om missende waardes op de X-as alsnog (leeg) te plotten. Andere grafieken hebben hier namelijk ook 'last' van.

LadyIT, 2017-01-01:
Lijkt me een nette oplossing!

dennissiemensma, 2018-03-05:
Dit specifieke probleem komt (naar mijn mening) te weinig voor om hier echt actie voor te ondernemen. Ik begrijp helemaal dat het mooi is om dit net te krijgen, maar gezien bovenstaande comment over mijn tijdsbesteding zie ik realistisch gezien geen oplossing gemaakt worden.

closed 2018-03-05 by dennissiemensma