dsmrreader/dsmr-reader #200: Na update naar 1.5 "Internal Server Error"

LadyIT, 2017-01-01:
Update naar 1.5 via ./deploy.sh (met gebruiker dsmr) krijg ik een "Internal Server Error" als ik de pagina wil openen. Opnieuw opstarten van RaspPi 3 heeft geen effect. In de logfile zie ik diverse errors voorbij komen (zie bijlage) . Wat gaat er niet goed? Thnx!
[updatecode_v1.5.txt](https://github.com/dennissiemensma/dsmr-reader/files/680129/updatecode_v1.5.txt)

dennissiemensma, 2017-01-01:
Ik zie geen `(dsmrreader)` vóór je terminal staan. Het lijkt erop dat die niet de goede omgeving (VirtualEnv) gebruikt daardoor.

Er staat nu: `dsmr@raspberrypi:~/dsmr-reader $`
Ik zou iets verwachten als: `(dsmrreader)dsmr@raspberrypi:~/dsmr-reader $`

Zie ook [deze comment](https://github.com/dennissiemensma/dsmr-reader/issues/186#issuecomment-269906177). Concreet moet je doen:

* `sudo su - dsmr`
* `source ~/.virtualenvs/dsmrreader/bin/activate`
* `cd ~/dsmr-reader`
* Als het goed is zie jet nu `(dsmrreader)` staan: `(dsmrreader)dsmr@raspberrypi:~/dsmr-reader $`
* Nu kun je updaten (gewoon `./deploy.sh` opnieuw uitvoeren).

Lost dat het op?

LadyIT, 2017-01-01:
Yes, werkt perfect. Dank! Ben nog lerende in deze wereld :-).

dennissiemensma, 2017-01-01:
Top. Het is goed om meteen even het laatste deel van hoofdstuk 6 van de installatie guide te doen: http://dsmr-reader.readthedocs.io/nl/latest/installation.html#virtualenv

Dan doet die dit voortaan automatisch.

closed 2017-01-01 by dennissiemensma