dsmrreader/dsmr-reader #202: Paar verzoeken m.b.t. notificaties

ThinkPadNL, 2017-01-01:
De notificatie mogelijkheid spreekt mij wel aan. 
Alleen dagelijks is nogal overkill en iets wat mij waarschijnlijk snel gaat irriteren. Is het misschien mogelijk om deze interval te vergroten naar maandelijks? (Om 0:00 / 0:01 van de eerste van de maand lijkt mij dan handig).

Ik zou dan ook graag de meterstanden van dat moment meegestuurd willen hebben.
Een mock-up van hoe ik het graag zou zien:

> **_Energieverbruik december 2016:_**
> **Laagtarief:** 78 kWh
> **Hoogtarief:** 89 kWh
> **Gasverbruik:** 56 m3
> 
> **_Meterstand op 1 januari 2017:_**
> **Elektra laagtarief:** 1234.567 kWh
> **Elektra hoogtarief:** 4567,891 kWh
> **Gas:** 7654.321 m3
Ik zag dat ook al ergens in een ander issue voorbijkomen. Maakt het wel makkelijk als je zelf nog e.e.a. met de meterstanden wil rekenen/ergens anders bij wilt houden.

Een tweede verzoek is om de mogelijkheid toe te voegen dat een notificatie per mail wordt gestuurd. Dan is het platformonafhankelijk. Al snap ik ook wel dat dat niet eenvoudig toe te voegen is, want de mail moet ergens verstuurd worden. Dan moet je al SMTP-instellingen e.d. doen

dennissiemensma, 2017-01-01:
Bedankt voor je verzoeken. Ik kan me zeker (in een deel) vinden. 

Het mailen is wel een technische uitdaging die goed getest mag worden. Mails zonder domeinnaam en SPF-records in de DNS worden vaak al gefiltered als spam. En voor thuis-Internet blokkeren de meeste providers ook al SMTP-poort 25 (en heb je inloggegevens nodig via de provider).

Zet dit voor nu op de lijst voor `v1.7` om er dan weer naar te kijken, gezien [`v1.6` inmiddels aardig vol is](https://github.com/dennissiemensma/dsmr-reader/milestone/25). De ticketlijsten voor de versies staan overigens nooit vast, helemaal niet aan het begin van een nieuwe release.

ThinkPadNL, 2017-01-01:
Het mailen is inderdaad wel een technische uitdaging. Maar als men dsmr-reader draaiend heeft kunnen krijgen, dan moet het opzoeken & instellen van de gewenste SMTP-instellingen niet zo'n punt zijn lijkt mij ;-) Wanneer men gewoon het eigen mailadres gebruikt dan komt het niet zomaar in de spamfolder. Heb voor monitoring in m'n thuisnetwerk ook m'n Gmail-adres ingesteld en dat werkt prima.

Het versturen van notificaties via mail lijkt mij wel een zinvolle toevoeging. Ik kan namelijk wel een paar tegen-argumenten bedenken voor de huidige manier waarop notificaties zijn geïmplementeerd:

- Prowl (iOS) is een betaalde app
- Er zal wellicht ook iemand een verzoek indienen om op zijn/haar Windows Phone een notificatie te kunnen krijgen, dat is nu niet mogelijk
- Wellicht zijn er ook mensen die dit gebruiken en geen smartphone hebben (bestaan die :D ?) en dus geen pushnotificaties kunnen ontvangen.

E-mail is dan wel vrij universeel in dat opzicht.

dennissiemensma, 2017-01-01:
Klopt, ik ben het daar helemaal mee eens hoor. De telefoon-notificaties zijn meer een nice to have, en optioneel. 

E-mail is zeker werkend te krijgen, het is alleen wel iets is wat goed getest mag worden zodat het een oplossing is die overal bij iedereen werkt. Altijd.
Uiteindelijk zie ik ook liever e-mailnotificaties dan via een (externe) app, omdat het zowel een externe app vereist en voor iOS betaald is (mocht je nog een andere gratis service weten dan hoor ik dat graag). Precies zoals je al aangeeft.

ThinkPadNL, 2017-01-01:
> mocht je nog een andere gratis service weten dan hoor ik dat graag
Voor m'n domoticasysteem ([Home Assistant](https://home-assistant.io/)) gebruik ik [Telegram](https://telegram.org/) om notificaties vanuit het systeem naar mijzelf te sturen (wasmachine klaar, plaatje als er iemand aanbelt e.d.). Je maakt dan een zgn. 'bot' aan en kan dan berichtjes sturen. Is vrij eenvoudig (gewoon HTTP-calls), zie ook de [documentatie](https://core.telegram.org/bots).
Werkt op [zo'n beetje alle platformen](https://telegram.org/apps).

Maar feit blijft dat je dan ook weer een app hebt, terwijl e-mail op elk device met internet werkt.

ThinkPadNL, 2017-01-01:
Ik heb dit issue even opgesplitst in meerdere issues (#205 en #206), dat maakt het voor jou wat overzichtelijker.

dennissiemensma, 2018-03-05:
Ik weet niet of je dit project nog steeds gebruikt (ik kan me goed voorstellen dat met jouw wensen het te verperkt is), maar realistisch gezien kan/ga ik het niet allemaal verwerken. Toch erg bedankt voor je al goede ideeen, die zijn/waren echt top!

closed 2018-03-05 by dennissiemensma