dsmrreader/dsmr-reader #203: One-click installer?

ThinkPadNL, 2017-01-01:
Het installeren van `dsmr-reader` omvat nu nog best wel wat stappen als je kijkt naar https://dsmr-reader.readthedocs.io/nl/latest/installation.html

Terwijl je eigenlijk de commando's 1 op 1 overneemt en de rest vanzelf gaat. Is het wellicht een idee om het in één script te zetten, zodat het voor een minder gevorderde gebruiker nog makkelijker/sneller te installeren valt?

Een goed voorbeeld is bijv. 'Pi-hole', een ad-blocker die op een Raspberry Pi draait en op die manier de ingestelde devices reclamevrij probeert te houden. Het is erg simpel te installeren, je hoeft namelijk maar één (!!) commando uit te voeren op de commandline: `curl -L https://install.pi-hole.net | bash`

Is natuurlijk een vrij smerige en gevaarlijke manier om een curl-opdracht naar Bash te pipen, maar het is wel erg gemakkelijk! In mijn geval heb ik tijdens de installatie van `dsmr-reader` voor de webinterface ook gewoon admin/admin gekozen, dat kun je later immers toch simpel wijzigen indien je dat nodig vind. Maar voor een standaard setup (PostgreSQL, P1 op `ttyUSB0` e.d.) is zo'n script best handig lijkt mij. De wat meer gevorderde persoon installeert alles toch met de hand, of past zelf het script aan.

dennissiemensma, 2017-01-01:
Bedankt voor de tip. Ik zal zoiets zeker overwegen.

In #33 heb ik ooit een soortgelijk iets voor ogen gehad, maar ik heb toen maar besloten mij te focussen op heldere documentatie. Inmiddels is die redelijk compleet en zou het een optie zijn om zoiets te maken.

Ook dit zet ik voor later op de planning om te kijken hoe en wat.

dennissiemensma, 2017-03-08:
Een deel is inmiddels al uitgeschreven in [xirixiz/dsmr-reader-docker](https://github.com/xirixiz/dsmr-reader-docker). (#273)

dennissiemensma, 2018-01-17:
Eerste opzet gemaakt, getest in een VM in Debian Jessie. Volgende stap is kijken of het op een echte Pi met Raspbian werkt.

dennissiemensma, 2018-01-18:
Ik ben aan het testen op Raspbian, maar die heet dezelfde issues met `locales` als Debian:

```
Error: The locale requested by the environment is invalid.
Error: could not create default cluster. Please create it manually with

  pg_createcluster 9.4 main --start
```

Dat vereist handmatig fixen van locales en ik wil dat liever niet automatiseren. Ik laat daarom het script zitten en beperkt de docs tot twee variant:
* Shortlist van alle commands (feitelijk het script).
* De bestaande, uitgebreide handleiding.