dsmrreader/dsmr-reader #204: Mindergas.nl API: Tijdstip van verzending

ThinkPadNL, 2017-01-01:
Ik zit even de code door te kijken en volgens mij wordt hier: https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_mindergas/services.py#L39#L46 het tijdstip van verzending gasmeterstand naar mindergas.nl bepaald. Als ik het goed zie, gebeurt dat om 0:00 exact.

Ik heb zo mijn twijfels of dat een handige keuze is. Mindergas.nl zegt zelf namelijk:

> Wanneer posten?
> Om de serverbelasting wat te spreiden is het aan te raden om een vertraging in te bouwen tussen het uitlezen van de meterstand en het opsturen naar de server. Bijvoorbeeld een vertraging van een willekeurig aantal minuten tussen 0 en 60.

Zeker als er meer en meer mensen via dsmr-reader hun gasmeterstand gaan uploaden naar Mindergas.nl, dan is het wel zo vriendelijk om daar wat willekeurigheid in aan te brengen (Meterstand uploaden tussen 0:00 en 0:30 ofzo). Anders is het een soort botnet-in-wording :P

dennissiemensma, 2017-01-01:
Dat klopt, je kunt het bij jezelf checken in de logs met:
`zgrep MinderGas /var/log/supervisor/dsmr_backend.log.* -B 10`

Het toevoegen van een willekeurige vertraging lijkt mij prima.

dennissiemensma, 2017-01-04:
Merged 0aab1828b1abf279cb145dc846e5fc8d2683ebbd to `development` branch.

closed 2017-01-04 by dennissiemensma

dennissiemensma, 2017-01-09:
Deze is bij toeval onderdeel geworden van de patch die net uitgebracht is. 

De eerstvolgende keer dat de export gedraaid wordt, pakt hij middernacht + 15 tot 59 minuten voor de **opvolgende keer**. Dus feitelijk die van overmorgen.

ThinkPadNL, 2017-01-09:
Goed bezig, heb hem geïnstalleerd. Zal het even controleren komende week met `zgrep MinderGas /var/log/supervisor/dsmr_backend.log.* -B 10`