dsmrreader/dsmr-reader #206: Uitgebreidere notificaties

ThinkPadNL, 2017-01-01:
Dit ticket is ook een beetje vanuit #205 ontstaan. Bij het versturen van een notificatie zou ik graag zien dat deze wat uitgebreider is dan nu het geval is. Een voorbeeld van hoe ik het graag zou zien:

> **_Energieverbruik december 2016:_**
> **Laagtarief:** 78 kWh
> **Hoogtarief:** 89 kWh
> **Gasverbruik:** 56 m3
> 
> **_Meterstand op 1 januari 2017:_**
> **Elektra laagtarief:** 1234.567 kWh
> **Elektra hoogtarief:** 4567,891 kWh
> **Gas:** 7654.321 m3

Het uitspugen van de meterstanden zag ik ook al ergens in een ander issue voorbijkomen. Maakt het wel makkelijk als je zelf nog e.e.a. met de meterstanden wil rekenen/ergens anders bij wilt houden.

2. Een tweede verzoek is om de mogelijkheid toe te voegen dat een notificatie per mail wordt gestuurd. Dan is het platformonafhankelijk. Al snap ik ook wel dat dat niet eenvoudig toe te voegen is, want de mail moet ergens verstuurd worden. Dan moet je al SMTP-instellingen e.d. doen

3. Een derde verzoek zou zijn om in bovenstaand voorbeeldbericht dan ook nog een kleine berekening t.o.v. vorige maand / zelfde maand in jaar ervoor mee te nemen, op deze manier:

> **_Energieverbruik december 2016:_**
> **Laagtarief:** 78 kWh (5% minder dan vorige maand)
> **Hoogtarief:** 89 kWh (10% meer dan vorige maand)
> **Gasverbruik:** 56 m3 (6% minder dan vorige maand)
> 
> ....

Al is dat misschien wel wat veelgevraagd. Maar ik vraag het hier toch maar, dan heb ik iig m'n idee gedeeld 😃

dennissiemensma, 2017-01-18:
Ik zie dat ik hier nog niet op gereageerd had maar ik vind het een goede toevoeging. Sowieso het opslaan van meterstanden zodat ze makkelijk getoond kunnen worden in het archief, of zoals je zelf aangeeft: in notificaties. Dit pak ik verder op in #243

Wat betreft het vergelijken met andere maanden, dat lijkt me ook zeker een mooie toevoeging. De exacte locatie(s) moet ik nog even over nadenken, maar in het kader van bewust zijn van energieverbruik lijkt me dat prima.

dennissiemensma, 2018-03-05:
Qua notitifcaties laat ik het voor wat het is. Ik zie hier echter nog wel toekomst in in ofewel Trends of in de Vergelijk-pagina (die nu wat simpel is).

closed 2018-03-05 by dennissiemensma