dsmrreader/dsmr-reader #207: Statuspagina denkt soms te gauw dat er geen data meer binnenkomt.

ThinkPadNL, 2017-01-01:
Als ik een aantal keren achter elkaar de pagina /status ververs, dan krijg ik soms wel eens de melding:
> Het is een tijdje geleden sinds het laatste telegram uitgelezen is. 
> Meest recente meting: nu

Als ik 15x op F5 druk, dan gebeurt dit zo'n 2-3x. 

Er zal waarschijnlijk een stukje logica achter zitten, wat nu te strak staat afgesteld? De meter spuugt elke 10 seconden een nieuw telegram uit, dan is het een beetje vreemd als hij na 10-15x op F5 drukken (vrij rustig, niet als een bezetene) die melding al geeft. Ik zou pas na 30-60 seconden geen nieuwe data vanuit de slimme meter die melding weergeven. Er kan immers ook door #188 een telegram zijn geweigerd, maar het telegram daarna wel weer OK zijn bevonden. Pas na één minuut gaan gillen lijkt mij dan beter, om minder paniek te zaaien.

ThinkPadNL, 2017-01-01:
Gelijk nog een vraag m.b.t. de statuspagina. Ik zie nu staan: `Onverwerkte telegrammen: 25`
Wat betekent dat? Ik heb wat in de code zitten zoeken en zie dat dit de NL vertaling is van 'unprocessed telegrams'. Helaas kom ik daar niet echt verder mee.

Het is groen omkaderd "dus het zal wel goed zijn", maar op mij komt het wel over alsof er iets mis is ("Waarom zijn de telegrammen niet verwerkt"). Ik heb dsmr-reader draaien in een VM, dus aan de resources kan het niet liggen zou je zeggen. Maar wat is het dan wel? Moet deze waarde nul zijn?

Misschien ook iets om op te nemen in de FAQ?

dennissiemensma, 2017-01-02:
Momenteel berekent de pagina [het aantal seconden sinds de laatste meting](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_frontend/views/status.py#L26) en toont vervolgens een rode melding [wanneer die leeg (geen data) of meer dan 60 seconden is](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_frontend/templates/dsmr_frontend/status.html#L61). Wellicht zit daar een foutje in. Ik kan de offset wel naar bijvoorbeeld 90 of 120 seconden zetten, al is het raar dat je die melding ziet wanneer alles gewoon gemeten wordt.

Wat betreft `Onverwerkte telegrammen`, het is een vrij technisch verhaal, maar ik vermoed dat je die ins en outs wel interessant vindt. Alle data wordt in drie stappen verwerkt:
* Telegrammen worden uitgelezen en opgeslagen (telegram onverwerkt). (`dsmr_datalogger` process)
* Synchroon worden de onverwerkte telegrammen behandeld en eventueel gegroepeerd per minuut (telegram is nu verwerkt). (`dsmr_backend` process)
* Na elke dag worden de verwerkte telegrammen geaggregeerd in uur- en dagtotalen. (`dsmr_backend` process)

Die laatste groep data wordt gebruikt in Trends, Archief, etc. Ik heb geprobeerd om die flow in de nieuwe Status-pagina inzichtelijk te maken, maar ik denk dat er nog eea aan uitleg bij mag dan.

Wat betreft jouw situatie, in de praktijk zullen er max 6 a 7 onverwerkte telegrammen zijn, wanneer je instellingen op 'groeperen per minuut' (standaard) staan. Het kan wel zijn dat het achtergrondproces bezig is met iets anders (export, backup, etc), waardoor de telegrammen tijdelijk niet verwerkt worden.

Je kunt het verwerken van de telegrammen volgen in Supervisor via `sudo supervisorctl` en dan `tail -f dsmr_backend` in Supervisor. Elke minuut zou je activiteit moeten zien.

ThinkPadNL, 2017-01-02:
Duidelijk verhaal 👍

ThinkPadNL, 2017-01-31:
Eigenlijk is het status blokje bij mij nooit meer groen. De melding die er staat 'Over 13 seconden' vind ik ook nogal vaag. Heeft hij nou data binnengekregen of niet ;) 
[image]

Zou dit komen doordat ik #187 gebruik?

dennissiemensma, 2017-01-31:
Ja, hij baseert dit op het tijdstip van het telegram. Ik heb nog niet eerder meegemaakt dat die voorliep.

Het lijkt erop dat het telegramtijdstip, die de slimme meter doorgeeft in het bericht, voorloopt op die van je server. Of dat je server achterloopt qua tijdstip.

Je zou kunnen kijken wat voor tijdstippen in de telegrammen staan in de logs van de datalogger en tegelijkertijd een `date` doen op de command line van je server.

ThinkPadNL, 2017-02-01:
De slimme meter lijkt ongeveer 16 seconden voor te lopen:
[image]

Maar ik heb NTP-synchronisatie al aan staan in die VM, dus ik weet niet zo goed wat m'n opties zijn:
```
(dsmrreader) dsmr@dsmrreader-vm:~/dsmr-reader$ timedatectl
      Local time: wo 2017-02-01 19:50:50 CET
  Universal time: wo 2017-02-01 18:50:50 UTC
        RTC time: wo 2017-02-01 18:50:50
       Time zone: Europe/Amsterdam (CET, +0100)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
(dsmrreader) dsmr@dsmrreader-vm:~/dsmr-reader$
```

Jij een idee?

dennissiemensma, 2017-02-01:
Ik denk dat je conclusie klopt. Vermoedelijk kan ik dit wel oplossen in de webinterface, door ook een negatieve delta toe te staan.

Voor nu hoef je je hier dus geen zorgen over te maken. De applicatie denkt dat de telegrammen in de toekomst zijn gemaakt, heel cool een futuristisch.

dennissiemensma, 2017-02-08:
@ThinkPadNL ik ben weer bezig met de volgende release en momenteel de Status pagina. Ik heb nu overal uitklap-linkjes geplaatst waarachter meer informatie te zien is.

Kun je aangeven of dit voldoende duidelijkheid geeft?

Zie hieronder eerst de ingeklapte versie en daarna de versie waarin alles opengeklikt is:

[image]

[image]

dennissiemensma, 2017-02-08:
Ik heb tevens in c1e1fbe69ca3bbc9a4e7a8b60ba3964fd289727f een fix gemaakt voor jouw situatie waarbij de slimme meter voorloopt qua tijd. Hij reset hem (visueel) dan altijd naar het huidige moment en zet de delta (voor de rode melding) op nul, waardoor die altijd groen is.

Hij klaagt dan als het goed is alleen nog maar wanneer hij echt achterloopt.

ThinkPadNL, 2017-02-09:
Ziet er goed uit! Ik ben pas volgende week dinsdag weer thuis en in de gelegenheid om het te testen, ik zal dan even kijken hoe het eruit ziet :)

dennissiemensma, 2017-02-09:
Oke bedankt voor de input. Dan merge ik dat terug naar de ontwikkelbranch voor nu. Je kunt het nu nog niet echt makkelijk testen. Wellicht wel wanneer de eerste release candidates er zijn, maar dat duurt nog even.

Ik laat de Status pagina dan verder zo, gezien jij de enige was met feedback hierover.

closed 2017-02-09 by dennissiemensma