dsmrreader/dsmr-reader #208: Notificatie als er al een tijdje geen data vanuit de slimme meter meer binnenkomt

ThinkPadNL, 2017-01-01:
Vanuit #207 bedacht ik mij ineens dat het handig zou zijn om een notificatie te krijgen als er al een poosje geen data vanuit de slimme meter meer is binnengekomen. Het zou zonde zijn als je de P1-uitleeskabel perongeluk los zou trekken en er pas na een maand achter zou komen.

Wat 'een poosje' is staat nog open ter discussie 😉 Een uur?

Een éénmalige notificatie is genoeg, je wilt niet dat de software je continu blijft spammen. Dat zou erg irritant zijn als je niet in de buurt van je slimme meter bent, om het probleem te kunnen oplossen.

Tenzij deze functionaliteit reeds geïmplementeerd is, dan mag dit issue worden gesloten 😆

goegol, 2017-01-01:
+1! goede feature!

LadyIT, 2017-01-02:
Zou tof zijn!

dennissiemensma, 2018-03-20:
Inmiddels werkend. Na een uur stuurt die een notificatie. Twaalf uur erna kijkt die opnieuw naar de status.

closed 2018-03-20 by dennissiemensma

dennissiemensma, 2018-03-21:
Uitgebracht in [v1.15](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.15.0).