dsmrreader/dsmr-reader #209: solar panel gross-production via Youless

mkruiver, 2017-01-02:
Dennis,

Thanks for developing DSMR-reader and the extensive documentation!
Request: Would you consider adding a module for importing data of gross-metering of solar panel production? Solar panel enthusiasts tend to separately monitor their gross production with a separate metering device and pulse counter (Youless, for example). It would be very neat if the DSMR-reader could also save the gross production data and plot that into a combination chart (generation & consumption).

Kind regards,
Martin

ThinkPadNL, 2017-01-02:
https://gathering.tweakers.net/forum/list_message/45838667#45838667 is probably what you are looking for. No need to reinvent the wheel again 😉

dennissiemensma, 2017-01-02:
Hi Martin, thank you for you kind words, it's appreciated. :]

Regarding you request (if I understand it correctly), the scope of this project is restricted to everything related to DSMR and the smart meter data itself. So I won't integrate external data into this (any more).

However, I do actively support requests to (automatically) export any data read by the application (as it's "your own" data) in any form (such as CSV or using an API). 
There is currently support for **MinderGas** and there will also be support for **PVOutput** (#9) soon. 

I have no experience with the Youless myself, but I see that [it also supports PVOutput as well](http://www.youless.nl/energiemanager.html), which seems a better platform for this anyway.

Will integrating #9 help you in any way? Please note the discussion there for any specific details.

mkruiver, 2017-01-02:
Thanks for your quick response. Effectively, implementation of #9 would also address this request of having both generation and consumption in one overview. I will follow #9, thanks for your dev work!

(Should this issue be considered closed / merged with #9?)

dennissiemensma, 2017-01-03:
Yeah, I think PVOutput will be a decent solution for this.

closed 2017-01-03 by dennissiemensma