dsmrreader/dsmr-reader #210: Automatic refresh of dashboard charts

kvolleberg, 2017-01-02:
Hi Dennis,
First off: happy new year! 

I have upgraded dsmr to the 1.5 version. You've made some nice enhancements. 
I noticed you updated the dashboard page. In my 1.4 version, I included some code to automatically update my electricity charts with the latest reading. I thought this was useful for realtime monitoring; this saved me some unneccessary F5'ing.

Now, with the 1.5 version, my local mods have been removed and, as things have been changed quite a bit, it's not a matter of copy-paste anymore. Plus: it's not that robust to do this locally.

Therefore: I would like to suggest an auto-update method in a next version. What are your thoughts on this?

dennissiemensma, 2017-01-02:
Happy new year to you as well!

I think it's a good improvement, so I'll surely add this later.

Regarding your local modifications, it's actually very easy to patch this feature in this version. This should do it already:

```
diff --git a/dsmr_frontend/templates/dsmr_frontend/dashboard.html b/dsmr_frontend/templates/dsmr_frontend/dashboard.html
index 8058e14..5ce2b45 100644
--- a/dsmr_frontend/templates/dsmr_frontend/dashboard.html
+++ b/dsmr_frontend/templates/dsmr_frontend/dashboard.html
@@ -365,6 +365,7 @@
                 /* Auto refresh. */
                 update_header();
                 setInterval(function(){ update_header(); }, 2500);
+                setInterval(function(){ update_graphs(); }, 5000);
                 
                 initialize_graphs();
             });
```

So juste add `setInterval(function(){ update_graphs(); }, 5000);` in `dsmr_frontend/templates/dsmr_frontend/dashboard.html` [around line `368`](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.0/dsmr_frontend/templates/dsmr_frontend/dashboard.html#L368) and save it. (5000 = milliseconds refresh rate)

A browser page refresh should suffice to apply the changes.

dennissiemensma, 2017-01-03:
Merged to `development` branch.

closed 2017-01-03 by dennissiemensma

dennissiemensma, 2017-01-09:
This ticket has been part of the new patch that was just released. It should update your dashboard graphs every few seconds. 

Please note that it may take a minute when you have enabled the 'group readings by minute' setting in the configuration.