dsmrreader/dsmr-reader #213: Grafiektype op dashboard aanpassen naar 24h/today so far?

ThinkPadNL, 2017-01-03:
Op de homepage ('Dashboard') staat de grafiek `Recent Elektriciteitsverbruik (Watt)`
Deze toont het laatste halfuur. Maar ik vind die grafiek niet zo handig eigenlijk. Zeker als je bijv. TV ofzo zit te kijken dan is het vaak een vlakke lijn met weinig wijzigingen. In het dashboard (InfluxDB als backend en Grafana als frontend) had ik een grafiek die 'today so far' toonde, [plaatje](https://tweakers.net/ext/f/8vIEnhEL7UHNhH6R786dViOy/full.png).

Ik zie liever een grafiek van de afgelopen dag, dan zie je bijv. ook snel hoe het stroomverbruik eruit zag toen ik aan het werk was. Nu moet ik dan een flink aantal keer klikken voor ik dat in beeld heb.

Het mooiste zou dan zijn als je een grafiek hebt waarin je kan zoomen, zoals hier: http://www.highcharts.com/demo/line-time-series

dennissiemensma, 2017-01-03:
Klinkt als een soortgelijke wens als in #189.

ThinkPadNL, 2017-01-03:
Dat lijkt inderdaad dezelfde vraag te zijn, ik ben dus niet de enige die dat mist 😄 
Het zoomen heeft een wat lagere prio, maar eigenlijk de gehele dag in één keer in beeld hebben is wel iets wat ik wat meer mis eigenlijk

ThinkPadNL, 2017-01-03:
Nog even gekeken, eigenlijk zoek ik eenzelfde soort grafiek als op /archive wanneer je een dag kiest.
[Deze](https://i.imgur.com/BCNPLXz.png) dus, maar dan vanaf 0:00 van vandaag, tot het tijdstip dat het nu is ('today so far' dus).

dennissiemensma, 2017-01-03:
Ja. Ik denk dat de huidige elektriciteitsgrafieken, op het Dashboard, hun beste tijd wel hebben gehad qua plotting.

In die laatste, in je initiele voorbeeld, zie ik ook wel een toegevoegde waarde (`Highcharts Demos › Time series, zoomable`). Ik zal tzt onderzoeken of die voor niet-commerciele projecten te gebruiken is en ook of die meerdere lijnen kan plotten.

ThinkPadNL, 2017-01-03:
Eigenlijk kan ik het heel kort samenvatten, de ideale grafiek zou net als op PVoutput zijn: [voorbeeld](http://pvoutput.org/intraday.jsp?id=38706&sid=35414&dt=20160330). De `fill area` ('energy') wil je het dagverbruik zien (hoogste meterstand van vandaag minus laagste meterstand), de rode lijn is het actuele verbruik ('+P' meen ik, in 'Watt').

dennissiemensma, 2017-01-04:
Ik ga al je feedback, samen met die van anderen, meenemen in een nieuwe opzet binnenkort. Dat betreffen dan vooral het Dashboard, Archief en bijbehorende grafieken/data.

dennissiemensma, 2017-01-17:
Om even alles te inventariseren (voor mijn overzicht) zijn er per onderdeel nu nieuwe verzameltickets. Voor dit ticket is dat vooral #235.

Daarin kijk ik in de eerste instantie wat er mogelijk is voor een eventuele proof of concept. Eventuele input of feedback kan dan later op gegeven worden. Niet alle wensen van iedereen zijn overigens doorgaans even haalbaar of praktisch.

ThinkPadNL, 2017-01-17:
Klinkt goed. Ik heb mij gesubscribed op dat issue en het andere issue waar je mij in had genoemd.
Ben heel benieuwd, hoor het graag als je iets klaar hebt om te tonen.

dennissiemensma, 2017-01-18:
Zal ik doen. Waarschijnlijk [gaat daar overigens wel enige tijd overeen](https://github.com/dennissiemensma/dsmr-reader/issues/9#issuecomment-273552910).

ThinkPadNL, 2017-03-27:
Enig idee wanneer je deze / #235 kunt oppakken? 
Tijdje zonder slimme meter gezeten (ben onlangs verhuisd), inmiddels weer eentje hangen en ontbreken van dit grafiektype is toch wel iets waar ik vrij gauw na installatie van `dsmr-reader` tegenaan liep. Ook #216 / #236 zou erg fijn zijn om beter inzicht te krijgen.

dennissiemensma, 2017-03-27:
Oorspronkelijk was het plan om deze in de v1.6 release op te pakken, maar ik liep tegen aardig wat uitdagingen aan in het afronden van stabiele ondersteuning voor DSMR v5. Daardoor heb ik v1.6 uiteindelijk uitgebracht met minder tickets dan ik had gepland, dus ook zonder deze. 

Voor v1.7 staat een stuk (backend) API (#230) op de planning en voor v1.8 wil ik graag weer frontend-verbeteringen gaan doen, zoals bijvoorbeeld dit ticket.

ThinkPadNL, 2017-03-28:
Jammer, nog even geduld hebben dus ;)

dennissiemensma, 2017-05-04:
Nog even een update. Gezien veel gebruikers de voorkeur hebben gegeven aan een PVOutput-koppeling (#9) en MQTT-ondersteuning ( #141), zal ik daar eerst mee bezig gaan (nu de API eindelijk klaar is). Dan weet je waar je aan toe bent.

dennissiemensma, 2018-03-05:
Zoals in een ander ticket beschreven, valt een (recente) frontend redesign mij wat tegen. Daarnaast heb ik je telkens 'beloofd' om er wat mee te gaan doen, maar dat is voor mij niet realistisch, helaas.

closed 2018-03-05 by dennissiemensma

dennissiemensma, 2018-07-11:
Inmiddels heeft de laatste release op het dashboard zoombare grafieken via #489.