dsmrreader/dsmr-reader #216: Grafiek stroomverbruik op archiefpagina klopt niet

ThinkPadNL, 2017-01-03:
I.v.m. #213 zat ik even naar de [grafiek van het stroomverbruik](https://i.imgur.com/BCNPLXz.png) op de pagina /archive te kijken.

Hoe langer ik er naar kijk, hoe meer vraagtekens het bij mij oproept.
Even een random grafiek van zonnepanelen [erbij](http://pvoutput.org/intraday.jsp?id=7664&sid=6485) zodat ik het wat beter kan uitleggen (ik heb geen zonnepanelen dus negeer het woord 'zonnepanelen', gaat puur om het grafiekje).

Je ziet in dat grafiekje de 'energy' (kWh) (die loopt lineair op, die wordt gedurende de dag steeds hoger en hoger en om 0:00 wordt hij weer nul en begint weer van voren af aan). Dit is bij de P1 eigenlijk de hoogste meterstand van die dag, minus de laagste meterstand van de dag. Hetgeen daaruit komt is je dagverbruik. Die waarde kun je bij elk telegram opnieuw berekenen. Als je weinig verbruik hebt dan stijgt de grafiek niet zo snel. Heb je veel verbruik, dan stijgt hij bijna lineair.

Dan is er nog de 'power', oftewel het actuele vermogen (Watt). Ik gebruik nu 400W, zet de waterkoker aan (1500W) waardoor ik op 1900W kom. Daarna slaat de waterkoker af en is het weer 400W. Een grafieklijn die gedurende de dag dus alle kanten op wiebelt.

Met dat in het achterhoofd heb ik eigenlijk geen idee waar ik naar zit te kijken als ik een willekeurige dag aanklik onder `/archive` en dan het elektragrafiekje bekijk.

ThinkPadNL, 2017-01-03:
Zoals ik het nu zie toon je de 'power' waarden (actueel verbruik) en moet de eenheid in de titel en legenda dus niet 'kWh' zijn maar 'Watt' (of 'W') en moeten de waardes ook `* 1000` omdat het nu onlogisch leest (0,082 kW == 82W). Ik vraag mij dan wel af wat voor waarde je daar toont. Als er om 10:00 een waarde van 0,082 staat, hoe moet ik dat interpreteren? Is dit een `avg` van het verbruik tussen 9:00 en 10:00 ? Of een `max` ?

Maar zelfs als je de eenheden aanpast vraag ik mij af of dit grafiekje nut heeft als je maar één datapunt per uur laat zien. Afhankelijk van je query (avg/max of iets anders) zou je dus een scheef beeld kunnen krijgen. Stel dat ik in een uur een continu verbruik heb van 400W en dan 20 minuten een dikke straalkachel aan heb staan, wat gaat dat ene datapunt per uur dan zijn? Volgens mij wordt je daar niks wijzer uit.

Ik zie dan liever die 'energy' zoals op PVoutput. Dan zie je namelijk tussen welke tijdstippen het verbruik het hardste steeg. Eventueel kun je daar het actuele verbruik (Watt) nog overheen plotten. Maar dan moet je het wel met meer datapunten doen (minstens elke minuut), anders heeft het naar mijn idee weinig zin om het lijntje te tonen.

P.S. Absoluut niet bedoeld om je af te kraken, maar als eindgebruiker roept dit vragen op haha 😉

dennissiemensma, 2017-01-03:
Geen zorgen hoor, dergelijke discussies zijn altijd goed. Vergeet niet dat ik dit project zelf al meer dan een jaar gebruik. En eigenlijk alleen maar kijk naar de dalen en pieken van dagen/maanden. De periode van 'dagelijks' kijken en zoomen op het verbruik ben ik al lang voorbij. Des te beter dat er weer verse input komt van eindgebruikers die graag van alles willen inzien (daar is het immers voor bedoeld).

Zoals [hier uitgelegd](https://github.com/dennissiemensma/dsmr-reader/issues/207#issuecomment-270006123) worden alle gegevens uiteindelijk geaggregeerd naar dag- en uurstatistieken. Het Archief, Trends, etc doet namelijk niets meer met de metingen zelf, alleen de aggregaties.
Deze bevatten geen Watt en ook geen metingen, maar [zijn gebaseerd op de meterstanden](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.0/dsmr_stats/services.py#L135) aan het begin en eind van elke dag of uur. Dus kWh.

* Dus wat je ziet is **geen actueel verbruik**, maar daadwerkelijke consumptie dat je uiteindelijk verbruikt en 'afrekent' (in kWh dus).
* De manier van plotten is niet cumulatief, maar simpelweg per datapunt.
* Wanneer je alle punten in de grafiek optelt zou je uit moeten komen op je dagverbruik.
* Je ziet dus eigenlijk ook tussen welke uren je verbruik het hardste steeg, maar niet cumulatief.
* `0,082 kW` om `10:00` betekent dus dat je [vanaf 10:00](https://github.com/dennissiemensma/dsmr-reader/blob/v1.5.0/dsmr_stats/services.py#L127) tot 11:00 `0,082 kW` heb verbruikt, op basis van je **meterstanden**

Wat betreft deze gegevens en het tonen ervan. Zolang #72 (retentie) nog niet uitgevoerd is en in werking is (eigen keuze voor gebruiker tzt), zijn er nog genoeg keuzes te maken, want de brondata is er dan nog. 

* Ik zie zelf wel wat zitten in het cumulatief tonen van de datapunten.
* Het wel of niet tonen van het actuele verbruik zal moeten samenhangen met of de gegevens door #72 wel of niet beschikbaar zijn.
* Hetzelfde geldt voor het wel of niet aggregeren van de datapunten naar kleiner interval dan een uur.

Ik zit er vooral mee met de afweging tot hoeveel details bewaard moeten blijven in verhouding met hoelang geleden je terugkijkt. Want ik zou mij volledig kunnen vinden in het bewaren van alle details (seconden of minuten) voor enkele maanden of zelfs een jaar, maar op den duur vraag ik me af hoevaak iemand terugkijkt in dergelijke details van bijvoorbeeld meer dan een jaar. Daaruit zijn dan ook de uur- en dagaggregaties uit ontstaan. 

Voorbeeld: Ik heb in 2015 tijdens de kerst heel veel stroom verbruikt (`19,74 kWh` op 1 dag!), maar zou ik nog moeten terugzien dat ik toen om `17:15:12` exact `7235` Watt verbruikte? Of is het voldoende dat het verbruik tussen `17:00` en `18:00` toen `4,2 kWh` is geweest?

En dit is niet alleen een punt van **willen** opschonen, maar ook omdat ik zie dat ik simpelweg na een jaar al een beste blob aan data heb en de applicatie er ook soms 'last' van heeft. Plus dat de schijfruimte op een RaspiberryPi kaart ook niet onbeperkt is (momenteel gaat dat nog wel prima).

En voor dat laatste verhaal is uiteraard wel een keuze of middenweg te vinden, maar daar mag goed over nagedacht worden.

ThinkPadNL, 2017-01-03:
Het begint mij al iets meer te dagen nu, dankjewel. 
Als ik alle datapunten in het grafiekje optel dan kom ik idd op het totale dagverbruik. Ik snap nu dus wat je doet in dat grafiekje: eigenlijk hetzelfde als op PVoutput, maar met idd het verschil dat jij het niet cumulatief doet. Het idee is zeker goed, maar door de grafiek cumulatief te maken (_de lijn stijgt dus alleen maar, enige wat verschilt is hoe snel hij stijgt tussen bepaalde uren_) wordt het voor veel mensen waarschijnlijk een stuk duidelijker waar ze naar kijken, dan op de manier waarop het nu geplot wordt.

Ik heb jouw huidige manier en de cumulatieve manier even tegen elkaar afgezet door alle datapunten even in Excel te zetten. Dat levert dan [dit plaatje](https://i.imgur.com/PjbrzOt.png) op.
De cumulatieve methode (oranje lijn) vind ik dan persoonlijk vele malen duidelijker. Voor de grafiek van het gasverbruik is dat hetzelfde.

Ik ben het verder helemaal met je eens m.b.t. retentie. Het is onnodig om het actuele verbruik (Watt) maanden of zelfs jaren te bewaren. Toch zou ik het zelf wel fijn vinden om het bijv. minstens één maand te bewaren. Ik keek met m'n vorige dashboard vrij regelmatig wat het verbruik tussen 0:00 en 6:00 was (sluipverbruik) zodat ik daar evt. [maatregelen](https://gathering.tweakers.net/forum/list_messages/1491207) tegen kon nemen als ik het te hoog vond. Ik merk dat ik die mogelijkheid in `dsmr-reader` nu wel mis, vandaar ook het ticket #213. Maar daarmee ben je er nog niet, dan moet ik al op dezelfde dag (voordat het 0:00 is!) die waardes bekijken. Een maand bewaartermijn van die data vind ik dan acceptabel, zodat ik het op m'n gemak via `/archive` nog eens kan terugkijken na een vakantie bijv.

dennissiemensma, 2017-01-04:
Zoals zojuist gecomment in ticket 213, ik ga alles binnenkort even opnieuw uitwerken zodat er één duidelijk voorstel is die (voor mij) technisch haalbaar is en door diverse actieve gebruikers even bekeken kan worden.

Ik zie alleen niet op (hele) korte termijn direct de mogelijkheid tot zo'n grote refactoring (want dat is er dan nodig). Hier zal wel even tijd overheen gaan, zodat je weet waar je aan toe bent. 

Dus wellicht dat je eventueel je vorige tooltje, tot die tijd, (ernaast) wil draaien als je echt deze data op korte termijn nodig heb. 
DSMR-reader heeft een API en [daarvan heb ik gisteren de dev docs bijgewerkt](http://dsmr-reader.readthedocs.io/en/development/api.html#script). Afhankelijk van hoe je vorig tool werkt, is dat wellicht op dezelfde manier aan te sturen. Ik draai hier thuis namelijk dat script en die stuurt twee installaties van dit project aan.

ThinkPadNL, 2017-01-04:
Bedankt, ik ben erg benieuwd!
Keep us posted als je updates hebt 👍

dennissiemensma, 2017-01-18:
Onderdeel van #236, deels gecombineerd met de input van 213.

closed 2018-03-05 by dennissiemensma