dsmrreader/dsmr-reader #217: Kan niet upgraden naar 5.1

eterpstra, 2017-01-04:
Hallo,

Ik draai momenteel v1.4 van DSMR Reader. Dit gaat perfect maar zou graag willen upgraden naar 5.1

Als ik inlog als dsmr user en de upgrade uitvoer dmv ./deploy.sh krijg ik de volgende error op het laatst:

[code]
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 295, in run
    requirement_set.install(install_options, global_options, root=options.root_path)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1436, in install
    requirement.install(install_options, global_options, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 672, in install
    self.move_wheel_files(self.source_dir, root=root)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 902, in move_wheel_files
    pycompile=self.pycompile,
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 214, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 204, in clobber
    os.makedirs(destdir)
  File "/usr/lib/python3.4/os.py", line 237, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.4/dist-packages/Django-1.8.16.dist-info'

Storing debug log for failure in /home/dsmr/.pip/pip.log
[/code]

Enig idee wat ik hiermee kan doen?

dennissiemensma, 2017-01-04:
Hallo @eterpstra bedankt voor het melden.

Waarschijnlijk mist het wisselen naar de VirtualEnv in de `~/.bashrc` van de `dsmr` gebruiker.

Wil je eerst dit handmatig proberen?
```
sudo su - dsmr

source ~/.virtualenvs/dsmrreader/bin/activate

cd ~/dsmr-reader

./deploy.sh
```

dennissiemensma, 2017-01-04:
Ik zal hier tevens een patch voor maken. Je bent niet de eerste die hier tegenaan loopt en ik kan bovenstaande stappen ook wel forceren in de deployscripts zelf.

eterpstra, 2017-01-05:
Hallo @dennissiemensma 

Dit was inderdaad de oplossing hiervoor! Thx!

Nu heb ik nog een vraag: Ik heb een clean image gemaakt op een andere pi. Nu wil ik hier de backup database gaan importeren. Kun je uitleggen hoe je dit kan doen? In de documentatie staat we hoe je het netjes kan backuppen, maar volgens mij nergens hoe je het terug kan zetten? (of zie ik iets over het hoofd?)

Alvast bedankt.

dennissiemensma, 2017-01-05:
Klopt, dit staat alleen nog in de documentatie van de ontwikkelversie, die overigens al wel in te zien is:
http://dsmr-reader.readthedocs.io/nl/development/faq.html#how-do-i-restore-a-database-backup

Je hebt het erover dat je een andere pi gebruikt als nieuwe omgeving. Wanneer je daar **geen** `./manage.py migrate` hebt uitgevoerd, hoef je de (lege) database daar overigens niet te verwijderen. Als je de complete guide hebt gevolgd, dan wel (hij moet compleet leeg zijn). 

Backups worden automatisch gemaakt, maar je kunt dus een directe backup forceren met:
`dsmrreader/provisioning/postgresql/psql-backup.sh` (staat [hier ergens tussen](http://dsmr-reader.readthedocs.io/nl/development/application.html#data-preservation-backups)).

Mocht je er desondanks niet uitkomen, ik hoor het graag. :]

eterpstra, 2017-01-06:
Thanks.

De backup die ik heb gebruikt was eerst nog van versie 1.4. Nu de upgrade bij mij wel was gelukt, kan ik dan de nieuwe backup (met v1.5.1) wel terugzetten naar een nieuwe instalaltie welke ook 1.5.1. draait?

De reden dat ik dit wil doen is omdat bepaalde apss corrupt zijn geraakt op mijn huidige pi. Ik wil dus een mijn schone image die ik klaar heb staan gebruiken om alles opnieuw te installeren en de sql backup dus terug zetten.

dennissiemensma, 2017-01-06:
Dat gaat goed hoor. 

Als je alles teruggezet hebt kun je de installatiehandleiding volgen. Op een gegeven moment komt er `./manage.py migrate` voorbij. Die upgrade oudere databases naar de versie die de code nodig heeft.

Mocht je alles al vóór de restore geinstalleerd hebben, dan kun je `./post-deploy.sh` uitvoeren als `dsmr`- user. Die doet hetzelfde (e.a.) en wordt ook gebruikt bij versie-upgrades.

dennissiemensma, 2017-01-09:
@eterpstra is dit gelukt?

eterpstra, 2017-01-10:
Hi Dennis,

Ik heb de upgrade kunnen uitvoeren. Alles draait nu op 1.5.1. Mijn ''schone image'' welke op een andere SD kaart staat,  draait ook op 1.5.1. Nu hoef ik in principe alleen maar de database terug te zetten toch ?

dennissiemensma, 2017-01-10:
Ja. Zorg er wel voor je de Supervisor-processen op de schone omgeving hebt uitgezet (`sudo supervisorctl stop all`) én dat je een lege database hebt daar. 

Je mag daar geen `migrate` hebben uitgevoerd. Heb je dat wel gedaan, maak dan een nieuwe database aan op de schone omgeving: 
```
sudo sudo -u postgres createdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
```

Tot slot kunnen alle processen daar weer aan als het goed is: `sudo supervisorctl start all`.

Bovenstaand verhaal staat ook hier: http://dsmr-reader.readthedocs.io/nl/development/faq.html#how-do-i-restore-a-database-backup

closed 2017-01-24 by dennissiemensma