dsmrreader/dsmr-reader #218: Grafiek Gasverbruik als staafdiagram weergeven

LadyIT, 2017-01-04:
Is het mogelijk om de grafiek gasgebruik in staafdiagram weer te geven? Dit leest wat makkelijker! Eventueel een weergave keuze menu toevoegen voor type grafiek.

dennissiemensma, 2017-01-04:
Bedankt voor de tip. Ik neem dit mee in de resign van de grafieken, samen met o.a. #213, #216 en #189 (en andere).

dennissiemensma, 2017-01-17:
Om even alles te inventariseren (voor mijn overzicht) zijn er per onderdeel nu nieuwe verzameltickets. Voor dit ticket is dat vooral #235.

Daarin kijk ik in de eerste instantie wat er mogelijk is voor een eventuele proof of concept. Eventuele input of feedback kan dan later op gegeven worden. Niet alle wensen van iedereen zijn overigens doorgaans even haalbaar of praktisch.

balk77, 2017-01-21:
hi, kun je ook de eenheid van gas verbruik corrigeren? Dashboard --> recent gas consumed, heeft tijd op de X as en kuubs op de y as. Het is onduidelijk in hoeveel tijd die kuubs zijn verbruikt. Elektriciteit is wel goed, J/s.

dennissiemensma, 2017-01-21:
@balk77 gasmeters geven hun meterstanden alleen elk uur door. Daarom zie je in de gasgrafieken puur het verschil in meterstanden tussen de meetpunten. De gasmeetpunten zijn in dit geval op hun kleinst **een uur**. 
Dit is wel veranderd bij nieuwe gasmeters en er komt nog ondersteuning voor die kleinere intervallen.

Wanneer er bijvoorbeeld op de gas X-as 18:00 staat dan is dat verbruik in het uur van 18:00 tot 19:00 gemeten. Of simpel gezegd, het verschil tussen de meterstanden op 18:00 en 19:00.

pyrocumulus, 2017-02-23:
Allereerst, heel gaaf project! 

Het zou wel fijn zijn als de resolutie van de gas grafiek ook verhoogd kan worden. Ik heb al een v5 meter hangen (gelukkig werkt dsmr wel gewoon) en zou daarmee dus veel accurater kunnen zien hoe mijn gasverbruik verloopt. Als je dan het verbruik per die vijf minuten zou kunnen zien en het cumulatief als staven in dezelfde grafiek (ongeveer wat je volgens mij van plan bent met kWh tegen W); dan zou dat heel veel informatie bieden.

Daarnaast, zou het ook fijn zijn als je terug door de grafiek heen zou kunnen gaan zoals dat ook kan voor de elektriciteit :) Ik vermoed dat het nut daarvoor nu nog beperkt is omdat je bij gas vooralsnog maar één datapunt/uur hebt maar met v5 is dat wel handig.

dennissiemensma, 2017-02-23:
@pyrocumulus bedankt voor de tips! 

Een deel (gasresolutie) heb ik inmiddels al klaarstaan via #221 voor een volgende release. Het scrollen door de gasgrafiek lijkt me ook zeker een nuttige toevoeging dan.

dennissiemensma, 2018-03-05:
Qua interface laat ik het nu voor wat het is. Qua DSMR v5 is er inmiddels afgelopen jaar al een release uitgekomen die een hogere resolutatie laat zien qua gas, als het goed is (ik heb zelf helaas nog v4).

closed 2018-03-05 by dennissiemensma