dsmrreader/dsmr-reader #219: Database backup herstellen (Postgres)

MWolthuis, 2017-01-05:
Hallo Dennis,

Allereerst bedankt voor deze software. Erg fijn om de energiegegevens lokaal en in eigen beheer te hebben.

Nu is mijn Raspberry Pi3 er alleen mee opgehouden en ben ik vanaf "0" alles weer aan het installeren/instellen. Het is mij alleen niet duidelijk hoe ik de gemaakte backup van de database weer kan herstellen.

Dit lees ik niet terug in de handleiding. Zou je mij op weg kunnen helpen?

dennissiemensma, 2017-01-05:
Goed om te horen! 

Het terugzetten van de backups wordt toegevoegd in de docs en is momenteel alleen beschikbaar in de ontwikkelomgeving-docs. Deze zijn overigens wel gewoon in te zien.

Zie [ook hier hoe en wat](https://github.com/dennissiemensma/dsmr-reader/issues/217#issuecomment-270724297) qua terugzetten. 

In de toekomstige versie van de docs staat het terugzetten [tevens beschreven bij de installatiestappen](http://dsmr-reader.readthedocs.io/nl/development/installation.html#database-backend), in de notitie-blokken.

MWolthuis, 2017-01-06:
Hartstikke bedankt! Appeltje, eitje inderdaad. Ik heb de ontwikkelomgeving-docs even niet zo snel gezien.

closed 2017-01-06 by MWolthuis